vitotai/BrewPiLess #206: Suggestion - Sanity check iSpindel data

2bbe, 2019-12-15:
Hi vito and thanks for the great work.
I do sometimes get invalid data from my iSpindel. See picture below.
Is it possible to add some "sanity check" for the data?
something like `if gravity < 0,5 use old gravity`

[image]

vitotai, 2019-12-18:
Low Pass Filter is sued for this issue. You can see the blue line wasn't drift too much.

Using this kind of "sanity check" is not as easy as you have thought.
What if we get reading before iSpindel is correctly placed into the fermenter or before it is "stable"? The correct readings will be rejected because they are out of ranges. Apparently, it won't work as expected unless other mechanisms involve, like checking successive readings. Then, how many  successive readings should be keep? Is it possible to have two or more WRONG readings?

2bbe, 2019-12-18:
I see your point.
Every time i get this "glitch" the gravity is reported as zero.
in reality, the gravity should probably be between 1.000 and lets say 1.200 for a really big beer
if this window is used during fermentation all unrealistic readings will be removed and the filtered gravity will be more correct.
if the first reading is outside the window, its probably a unwanted reading and could be discarded.

just my thoughts at this point