vitotai/BrewPiLess #207: Fan output occasionally stops working until device reapplied (without change)

fnivert, 2019-12-17:
Hi,
I just started my first tests using BrewPiLess with a NodeMcu. Thanks for a wonderful little design!

I use the FAN output configured on the D4 pin. It normally works as it should (or function between heater and cooler). But, sometimes I found the heater activated but not the FAN output. When in this state I go into the Setup page and just press Apply on the FAN device, i.e. no changes is made, and voila, the fan output goes active again.

Is the configuration of the FAN output lost in some way? Anyone else seen such behavior?

vitotai, 2019-12-17:
I've never used FAN. Frankly speaking, I've never noticed that there is an option for FAN.
According to the source code, fan is turned on whenever heating OR cooling is turned on. Other than turning on/off, the device is processed as other devices, like cooling and heating. I don't have any idea why this happened.

splitice, 2020-03-31:
I've found this to happen with cooler and heater too. Re-applying the device fixes it.

@vitotai any chance this has been fixed in the past 4 months?

vitotai, 2020-03-31:
I don't know. 
I've tried latest version, and it can be saved and restored successfully.

splitice, 2020-03-31:
I can replicate my issue it seems.

1. Have Beer Temp Limit set
2. Be outside parameters (i.e will heat or cool as soon as start)
3. Power-cycle.
4. Will activate heating / cooling but pysical GPIO will not change. Toggling inverted / not-inverted on the devices page will fix it, and further transitions work fine.

I am using an Auxiliary input through the ispindel interface. Might be related.

I'll upgrade the software as soon as this brew is done. My build is from Jan 17 so perhaps it's already been fixed.

vitotai, 2020-04-01:
Nothing has been done for this issue. 
This issue seems like something goes wrong during loading of hardware configuration of the fan. However, mine works so it's difficult to resolve it.

splitice, 2020-04-01:
@vitotai is there somewhere in particular I should look? or something you would like me to do as a test? Gut feelings?

vitotai, 2020-04-01:
Let me clarify this.
Do you mean that the fan configuration exists but doesn't work properly?

splitice, 2020-04-01:
I think the issue is lower level, since I'm able to replicate something similar with both HEAT and COOL outputs.

I don't have a fan configured.

vitotai, 2020-04-02:
> I think the issue is lower level, since I'm able to replicate something similar with both HEAT and COOL outputs.
> 
> I don't have a fan configured.

Sorry, I didn't notice that you have different issue with fnivert.
I would suggest to replace the relay module. Most relay modules I could find are of 5V specification. While most of them work under 3.3V, some of them don't. I have a couple of modules that don't work with 3.3v signal.

splitice, 2020-04-02:
I can confirm it's not the relay module, I stuck a multi-meter across the side DC last time it happened.

vitotai, 2020-04-12:
How about changing a NodeMcu/D1 mini/pro?
I would never say my software is perfect, but the part of code that controls the relay comes directly from original BrewPi. Over thousands of people have used it, and it seems to work fine for them.
On the contrary, a lot of bad knock offs of D1 mini/NudeMcu have been found to be faulty.

fnivert, 2020-04-13:
I do not longer use the fan output, only heater and cooler and one temp sensor. Today I again run into the state where the Brewpiless should turn on heater, but nothing happened. It said Heating but the heater output was not on. Via the Setup page I changes the heater output (Device 2, D4) to first "Not Inverted" (nothing happens), then back to "Inverted" and then the heater went on.

Do you think this is a hardware issue with the NodeMcu?

vitotai, 2020-04-15:
IMO, it is more like hardware issue than software.
As I've mentioned, that part of code has been used by thousands of people and worked fine.

I had similar problem but it was on my BrewManiacEx setup. Rewiring seems to fix it, although I still don't know why it suddenly didn't work before it was working fine.

splitice, 2020-04-15:
@vitotai Mines a LOLIN D1 mini V3.1.0. I've got 2 (one BrewPILess) the other one uses GPIOs without issue (led control). I'm skeptical of a hardware issue since it always works on further toggles - it's only ever the initial GPIO toggle that fails (fixed via setup page).

Given that I'll double up the toggles and see if that helps. If it's a hardware defect (or software integration issue) then that would fix it. If it's something more (e.g the GPIO write routine not being called) then it won't.

fnivert, 2020-04-15:
I have now reconfigured the Heater to be on Controller pin D7 which seems to be more standard. It has worked for 2 days now.

splitice, 2020-04-25:
@vitotai I did some testing and I have only been able to replicate the issue with a External Aux beer temp sensor.

When using the fridge probe I can't replicate.

ashgillman, 2020-05-08:
I was having an issue very similar @splitice , I think it may be because I was trying to have 2x output devices assigned to one function.

splitice, 2020-05-09:
I'm still investigating myself. But this is why it's a huge issue.

Power went out during the night.

[image]

Cooling never re-enabled. Requiring toggle from UI to activate.

vitotai, 2020-05-16:
I could help only when I can reproduce the issue by myself. The information you provided is not enough. It's like " mine works fine so far, and a lot of people's work fine, too". You can't get any help form the description.
If you think it's software issue, you might try other software, like Thorrax's or other software that controls Relay, like Tasmota. If they works well, then it should be software issue, otherwise it is hardware issue. Even if it's software issue, I might not be able to fix it. I've mentioned previously, I had similar issue and solved by rewiring.

splitice, 2020-05-16:
@vitotai I'm still testing on my end. Unfortunately my controller is under heavy usage and I've been unable to devote enough time to figure out the code in this project.

I can be certain that it is not hardware related. This module has been previously used for other applications and is not super cheap.

I've been able to verify at-least one easily testable way to trigger a failure to output situation already - the method proposed by @ashgillman. I'm fairly certain this is not the same bug as what I saw as AFAIK when doing the testing in April I had only one output (however I can't say with 100% certainty).

splitice, 2020-09-15:
I've been able to confirm that there is no issue when only one device of a type is defined. The problem stems from having 2 (or possibly more?) devices defined with the same function on different pins.