vitotai/BrewPiLess #210: error in grunt i18n

petobena, 2020-01-04:
Hi vito,

when I use "grunt i18n" in terminal, I become:

> peto@peto-server:~/Atom/BrewPiLess/htmljs$ sudo grunt i18n
> 
> [sudo] heslo pre používateľa peto: 
> Running "copy:jsfiles" (copy) task
> Copied 19 files
> 
> Running "processhtml:dist" (processhtml) task
> 
> Running "htmlmin:dev" (htmlmin) task
> Minified 16 files (2 failed)
> 
> Running "sass:dev" (sass) task
> 
> Running "postcss:dist" (postcss) task
> >> 7 processed stylesheets created.
> >> 7 sourcemaps created.
> 
> Running "comboall:main" (comboall) task
> Warning: Unable to read "build/js/vendor/dygraph.css" file (Error code: ENOENT). Use --force to continue.
> 
> Aborted due to warnings.
> peto@peto-server:~/Atom/BrewPiLess/htmljs$ 

can you help me (2 failed in htmlmin:dev and a warning)?
Thank you

vitotai, 2020-01-05:
Some files are missed. Please get the latest 4.0pre again.

petobena, 2020-01-05:
@vitotai I downloaded the last version, but still have the error

closed 2020-01-05 by petobena