vitotai/BrewPiLess #214: Yaler support (access over the internet behind NAT)

lukz, 2020-01-22:
Hi

I looking for a way of having brewpiless accessible over the internet. The problem is that my provider doesn't provide me with public ip address so I can't achieve this with simple port forwarding.

Solution I've found is to use Yaler (https://yaler.net/). It's a service similar to Ngrok that provides tunnel to specific port on your machine with publicly accessible address but they provide ESP8255 implementation (https://yaler.net/esp8266).

I was trying to integrate it to BrewPiLess. They simply suggest to replace:
`WiFiServer server(80);`
with
`YalerESP8266WiFiServer server("try.yaler.io", 80, "RELAY_DOMAIN");`

Although only place I could find WiFiServer was here https://github.com/vitotai/BrewPiLess/blob/master/src/BrewPiLess.cpp#L1451 and it's behind "ESP8266_WiFi" that I believe is off so looks like this code is not used.

Is AsyncWebServer used in BrewPiLess to create a web server for serving the interface? To make an implementation I guess I would have to add some changes to this server initialization. Where can I find this code?

allthepies, 2020-01-22:
Can't you just use a Dynamic DNS service and local port forwarding ?  That's what I do to expose my 
 BrewPiLess setup on the internet (also have an Alexa skill which can control BPL :) )

lukz, 2020-01-22:
As far as I know Dynamic DNS services only works if you already have public IP address, this only helps with problem of dynamic ip that's changing each time you connect to the internet.

allthepies, 2020-01-22:
I'm not sure how you can use the internet if you don't have a public IP address.   Your provider may choose not to expose that address to you and it may change over time but you will have an IP.  Or are you using some non-standard VPN based setup ?

What does https://www.whatsmyip.org/  report ?  Presumably it shows an IP address.

lukz, 2020-01-22:
I'm using the internet behind my provider public ip address. It's the same situation as you have in your home - probably only your home router have public ip address and other devices use are able to use internet without using their own public ip (they have local ip within your lan provided by router not visible from outside). Whatsmyip for those devices will display ip address of this router.

In my situation whatsmyip displays ip address of one of my provider devices.

vitotai, 2020-01-22:
First of all, I agree with allthepies. DDNS is easier and maybe safer. 
The WiFiServer is used by Thorrax and not used by BrewPiLess. 

I don't think it would be easy to integrate Yaler to EspAsyncWebServer. You can try, though, by implement the protocol over EspAsyncTcp.

lukz, 2020-01-22:
Yup, but DDNS doesn't solve my issue. They only provide domain for dynamic public ip addresses. I down own public ip address. 

I'll try to integrate Yaler into EspAsyncWebServer. Is this the place where interface server initialization happen? https://github.com/vitotai/BrewPiLess/blob/master/src/BrewPiLess.cpp#L1648

tamberg, 2020-01-22:
Hi @lukz, thanks for considering Yaler (I'm a founder). As far as I can tell, the current [ESP8266 library](https://yaler.net/esp8266) won't work here. If you still want to give porting a try, here's the [protocol documentation](http://s3.yaler.net/docs/YalerProtocol.pdf).

lukz, 2020-01-23:
Hi, thanks for the docs, I'll give it a shot.