vitotai/BrewPiLess #217: Active low relay module?

Naesstrom, 2020-02-20:
I have a dual relay board where both relays turns on at boot and stays on. Measuring the input lines both show 155mV  when off and ~3.3 when they are supposed to be turned on but as I said, they stay on all the time!
Is my relayboard shot or is there a setting I'm missing somewhere? Tried both inverted and non-inverted in the setup.

vitotai, 2020-02-21:
You can try using a wire to connect the input of the relay module to Ground and 3.3v to check how it works.
I had similar issues, and replacing the relay module solved my problem.

Naesstrom, 2020-02-22:
Yeah seems to be the relay module, replaced it with another one from the same batch I ordered but on that the relays are "stuck" and I have to tap them for them to turn on. Ordered new ones now!

closed 2020-02-22 by Naesstrom