vitotai/BrewPiLess #218: Attaching pressure sensor

Naesstrom, 2020-02-22:
I bought a 5V pressure sensor from [Banggood](https://banggood.app.link/jRLoWoLri4) that max on 30PSI. Read to late tough that I also should use a ADS1115 breakout. Is there something else I need, I checked the adafruit site and it says

> The absolute maximum analog input voltage is VDD + 0.3v. To avoid damage to the chip, do not attempt to measure voltages greater than VDD.

So by running it on 5V everything should be just fine?
[image]

vitotai, 2020-02-23:
I guess it should work. However, I have not setup my pressure sensor with ADS1115.

Naesstrom, 2020-02-23:
How did you connect it, I wired mine up to +5, gnd and GPIO36 but read that it might not work with wifi enabled.

vitotai, 2020-02-26:
I haven't tried it on ESP32 yet.

Naesstrom, 2020-02-26:
but on a esp8266? then you connect it just on the analog pin?

vitotai, 2020-03-04:
Yes, but It didn't work well. The error is significant. It might be good enough for monitoring, but not for controlling.
ADC via ADS1115 is implemented, but I haven't finished hardware setup, yet to test.

Naesstrom, 2020-03-04:
Cool, I'll order a ADS1115 and see how it works!

vitotai, 2020-03-16:
FYI, I just tried it on ESP32 and fine tuned some parameters. ADS1115 was also verified and fine tuned in latest update.

Naesstrom, 2020-03-21:
Great, waiting for my ADS1115 yet but looking forward to it!

Scales82, 2020-06-14:
@vitotai How is the ADS1115 wired to the esp8266? Im unfamiliar with them.

Naesstrom, 2020-06-20:
@Scales82 look at the picture in my first post in this issue

Scales82, 2020-06-23:
Thats an esp32 though isnt it? I assume the wiring is the same but using the appropriate SCL SDA pins on the esp8266? Also read grounding the address pad on the ADC is recommended to set the port address?

sedgington, 2020-06-28:
I'm having a develish time programming the Hiletgo ESP-32S with the wroom chip shown above. The espressif flash download tool says it runs and finishes when I flash the BrewPiLess.v4.0r1.esp32.bin, but when I repower the board I get none of the normal brewpiless AP as on the 8266 board. Is there anyone who can point out what I'm doing wrong or directions for doing it correctly? My objective is to get pressure spunding working with the ADS1115 board. I can find no wiring diagrams for the 8266 with this board for this purpose although under "pressure" brewpiless specifically asks if you are using this board. I'm assuming the wiring is not the same as the ESP32, and so would appreciate it if anyone could point me in the right direction about how to use the 8266 with the ADS1115 board.

Scales82, 2020-07-05:
@sedgington 
Although I am not familiar with what hardware you are using (I am using a wemos D1 mini). I believe that the wiring method is the same across all boards. The ADS1115 is an I2C device. so as long as you are supplying it with power, gnd, and have the analogue input going into A0. You should be able to connect the SCL and SDA pins to the I2C bus on your appropriate hardware and it should work (similar to connecting the I2C oled screen, you should be able to connect ADS to the same pins). The picture above is showing a esp32. My device uses slightly different pins for the I2C bus... I would suggest looking into what pins your device uses.

The ADS1115 should default to the appropriate I2C address, but I also connect my ADDR pin to GND to make sure it is.

Scales82, 2020-07-05:
@sedgington I just re-read your comment. It looks like you arent even getting the BPL to boot, I would suggest the ADS is not the problem and you are flashing incorrectly or using the incorrect FW. I use nodemcu flasher, not sure if that is compatible but maybe try that?

sedgington, 2020-07-05:
Thanks for your comment. I figured out what the problem was and made a long documented comment about how to fix it here:
https://github.com/vitotai/BrewPiLess/issues/128
If you scroll to the bottom or search on sedgington. 
For some reason Nodemcu works great for me with the ESP8266 chip, but not at all with ESP32. Go figure...

sedgington, 2020-07-07:
I just wanted to report back tht I have successfully implemented spunding with an ESP32 and it works flawlessly within a 1 psi margin. Since I used a solenoid that is capable of opening and shutting up to 5 x per second and controlled it with an IRLZ44NPbF logic level transistor, I'm wondering if it is possible to modify the script so it could check the status more rapidly once the pressure exceeds the setpoint. This might be a solution for the "pin hole" or "needle valve" requirement for the release of pressure as the solenoid could open and close rapidly until the pressure is reduced to the pressure required.
In any case, here is what I used to get this to work:
Solenoid: https://www.amazon.com/gp/product/B07XB6LF49/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
ESP32: https://www.amazon.com/gp/product/B07DBNHJW2/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1
IRLZ44NPbF: https://www.amazon.com/gp/product/B01ARKPMPK/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
Flyback Diode: https://www.amazon.com/gp/product/B01LW6N0AD/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1
Pressure transducer: https://www.auberins.com/index.php?main_page=product_info&cPath=38&products_id=311
1. I wrote up how to get the ESP32 to load brewpiless above in #128
2. I put a Flyback Diode (1N4001) across the positive and negative leads going into and out of the inductive load of the solenoid (light colored band on what you designate as the positive lead). This will reduce electrical spikes caused by the magnetic field in the solenoid when it turns on and off and protects your ESP32
3. Left to Right on the IRLZ44NPbF (with the top tab lying flush to the work surface) pin 1 is the Gate, pin 2 is Drain, and pin 3 is the  Source. The whole idea is that when a positive voltage (of less than 5 V) is applied to the Gate it allows voltage to flow from the Drain to the Source allowing the solenoid (mine is normally closed) to open. The trick is that when the voltage is not applied to the Gate something needs to pull the voltage low so that it turns off the flow from Drain to Source. This can be as simple as touching it with your finger. However, my ESP32 using pin D19 pulled it low automatically when no voltage was applied. One can also use a resistor to ground on this pin to assure this.
4. Because my solenoid is 12 V, one needs to have two power supplies to get this to work: A 12 V for the solenoid and a 5 V USB source for the ESP32. I used this breadboard design (without the button or the resistors, thanks to Martyn Currey) to lay this out. It is important that the grounds from the ESP32 and the 12 V power supply are connected:
http://www.martyncurrey.com/controlling-a-solenoid-valve-from-an-arduino-updated/
5. Once I had the the electrical design worked out so that the MOSFET controlled the solenoid, I went to Brewpiless and set it up with my local network. I then went to the Device tab and init/erase the EEPROM and then "refresh device list" For Device 0  I selected Device slot "3" which was GPIO19 corresponding to D19 on my ESP32. I selected the pin type as "inverted" and the function as "capper" and then hit the "apply" button.
6.Next I went to the "pressure" tab and set up the values as: Enable Pressure Transducer: on; Use ADS1115: on; ADS1115 Gain/Max input voltage: 4.096V; Control Pressure:on and then ran the Pressure Conversion parameters by following the "calibrate" dialogue. I used a 5 gallon empty Corny Keg charged with 11 psi CO2. First I set the value when the guage on my spunding valve read 0 and then when I attached it to the Corny Keg and it read 11 psi. I was happy to see that brewpiless reported on the Graph page that the pressure was 11 psi.
7. Finally, I went to the "Control" tab and under Capping Control at the bottom of the page selected the "Manual" tab. I turned on "Spunding" and then put the value as 12 psi (and hit "Apply". It is interesting to note that even though the read out is only showing whole numbers, because the actual pressure in the tank was 11.4 psi, if I changed this spunding number to 11 (and hit apply) the valve would immediately open. This is how sensitive the pressure transducer I used in combination with brewpiless is and is the reason for my suggesting that the solenoid could be rapidly opened and closed to reduce the pressure instead of reling on a specific diameter hole or setting on a valve as the controlling method.
Anyhow, hope this helps someone set this up. And, if anyone has found a 1/4" NPT pressure transducer that maxs out around 32 psi and gives a read out in the range of 0 to 5 V that is less expensive than the one I used, I would be delighted to hear about it. I spent some time searching and have not yet found one. Many thanks to Vitotai for creating such a great program.

vitotai, 2020-07-08:
First of all, we don't have an interrupt when the pressure is higher or lower than setting. (ADS1115 seems to have such feature, though. However, one additional PIN is necessary for it.) Polling is used to read the pressure values. I've tried reading as frequent as possible but found that the loading seem to be too high that some functions behave abnormally. Furthermore, quickly switching the solenoid and the relay seems to be a bad idea. Therefore, the minimum period to read pressure is 1 second, an arbitrary but reasonable value.

sedgington, 2020-07-08:
Hey thanks for your reply. Since I'm using a MOSFET to control the solenoid, rapid switching off and on is not a problem. Some projects use these as PWMs to control things like fan motors. Since my solenoid says it is capable of turning on and off 5 times per second, I was only trying to introduce a concept based on my hardware setup. If you could point me to where the code is that controls the ADS1115 and also the polling of the pressure in brewpiless, I could sniff around to see if there is a possible solution for my existing setup based on other Arduino projects. In any case, thanks again for all your great work.

vitotai, 2020-07-14:
You might want to check PressureMonitor.cpp, looking for loop(). Good luck.

sedgington, 2020-07-16:
I just did some simple tests with my solenoid valve: with or without a needle valve to control the flow of CO2 out of the solenoid. In both cases, the pressure dropped 1.4 psi before shutting off the solenoid (10.1 psi to 8.7 psi in this case, when the setting in BPL was 9 psi). So it seems like the needle valve is an unneeded extra cost: It just slows the release of CO2 and has no impact on the actual drop in pressure. Since a drop of 1.4 psi is entirely satisfactory from my point of view (bringing the pressure setting to around 3% below the setting requested), it seems unnecessary to fiddle further with the precision of measurement and release I was suggesting above. Just thought I would let you know how accurate the system is with my setup. Thanks for implementing this. 
I guess the only other thing you could do with the ADS1115 is to read 4 separate pressure transducers on 4 channels and then theoretically trigger 4 separate solenoids based on the BPL code. This however would likely require some major rewriting to coordinate the channels to the appropriate pin for the solenoid.

Scales82, 2020-11-04:
@sedgington 
I got a transducer for cheap. it was 1/8NPT though. But I just put a 1/8 to JIC fitting on it to accept a duotight connector and was all systems go.
https://www.aliexpress.com/item/33059909689.html

Naesstrom, 2020-11-09:
> @sedgington
> I got a transducer for cheap. it was 1/8NPT though. But I just put a 1/8 to JIC fitting on it to accept a duotight connector and was all systems go.
> https://www.aliexpress.com/item/33059909689.html

@Scales82 
I have the same transducer at home but havn't connected it, could you explain or take a picture how you connected it, googled JIC fitting but that didn't help me much 🤣

Scales82, 2020-11-09:
[image]

@Naesstrom 
Thats how i got it there... So in left-to-right sequence it is.
Sensor. 1/8NPT female to 7/16 JIC. (also known as MFL fitting in homebrew world). Duotight MFL fitting (Kegland). Hose...

jaogabrielpc, 2021-08-16:
Pessoal, alguem pode me ajudar a como ligar o sensor de pressão + ADS1115 no ESP8266 D1 mini?

Guys, can someone help me how to connect the pressure sensor + ADS1115 on ESP8266 D1 mini?

Naesstrom, 2021-08-17:
> Guys, can someone help me how to connect the pressure sensor + ADS1115 on ESP8266 D1 mini?

Check the image in the first post, it shows a ESP32 but connect it to the same pins on the D1 mini

Duncbayne, 2023-04-11:
Hi I've got a Node MCU ESP 8266 which I wanted to set up for pressure monitoring and autospund.

The pins are named differently to those on the ESP32 at the top of this thread it doesn't have D21 or D22.
[image]

Currently I have other ESP 8266 working for temperature control but have a spare to set up for pressure monitor and manage, or can I eventually incorporate it all on one unit?  Thank you if anyone is out there.

Duncbayne, 2023-04-11:
My next question is if this works I want to add a 12V solenoid for spunding control. 
The circuit diagram using mosfet linked to by @sedgington in post above shows the left mosfet pin going / from D2, with the esp8266 this is being used for SDA but @vitotai indicates to use pin D0 in the instructions about autospunding.
Am I correctly interpreting this?

Duncbayne, 2023-04-13:
Some progress wired up the pressure sensor but it only seems to work for about 15minutes at most. I notice it is a 5 to 12 volt input and wonder whether it is a little underpowered with the 5V from the supply which also goes to the esp8266.
Should I run it off the 12V intended for the solenoid or do I need a resistor to pull it down or up?