vitotai/BrewPiLess #219: Relay wiring diagram

ilaborde, 2020-02-26:
Hey I've been trying to make it work my brewpiless, but the relay not responds as expected. I'm not good with electronic, so I've tried with two different relays, thinking that the first one could be broken.  but same issue.
Electricy does not  flow to the sockets. 

Here is my setup: the blue wire, is the electricity and the red one goes to the socket

[image]

And my piless config:
[image]

The closest to works well, was with another configuration in the relay, and using "not inverted" option, but when the desired temperature was achieved the relay is still allowing the flow of electricity.

Any advice?

vitotai, 2020-02-26:
I am not good at electronics, ether. 
The only advices I could provide are
1. using a multi-meter to check if the PIN, say D5, behaves as expected.
2. checking the RELAY module by providing 3.3V/GND.
I have defected relay modules as other people do.

ilaborde, 2020-02-26:
could you upload a picture of your wiring please?

Hwerow, 2020-02-27:
Most relays need the Inverted setting

ilaborde, 2020-02-27:
Yeah, I've tried, but without success. Tomorrow I'll try with a diff relay, with an additional pin for 5v

Naesstrom, 2020-02-28:
Can't see much from that image but here's one of my latest build.
[image]
Brown are the "live wire" and blue is "neutral". from the middle (com) port on the relays they are connected to the heater and cooler and the NO is from the input part.

ilaborde, 2020-02-28:
Thank you so much, and how do you use your relay? just wired to nodemcu?

Naesstrom, 2020-03-01:
> Thank you so much, and how do you use your relay? just wired to nodemcu?

yup, followed the instructions here and wired it to D5 and D7
https://github.com/vitotai/BrewPiLess/blob/master/doc/HardwareExample1.md

ilaborde, 2020-03-02:
It's working, I'm using an old relay,  and I'm providing 5v removing the jumper and is working well.

Thanks so much

closed 2020-03-02 by ilaborde