vitotai/BrewPiLess #221: [Question] BrewNCalibrate just once or every brew?

Naesstrom, 2020-03-02:
No issue but rather a quick question. Doing a BrewNCalibrate with the iSpindel currently but should I do the same every brew or is the equation saved in BPL so I just need to do it the first time?

vitotai, 2020-03-04:
Brew&Calibrate was implemented because the center of gravity of my iSpindel changed every time when I charged it and unplug the USB cable.
You can "save" the formula and use it after if the center of gravity doesn't change a lot.

Naesstrom, 2020-03-04:
So like this then
1. Start log with brew and calibrate
2. After brew is done, check formula by pressing _f_ button
3. Copy to safe place
4. Add formula to **Gravity** page and enable **Calculated by BPL**

A great feature would be if BrewNCalibrate is enabled that it updates the formula on the gravity page automatically :D
--

vitotai, 2020-03-04:
It does. The formula is updated when you open the browser and update the data. The calculation is done by the computer or phone that running the browser.
However, you better save it in case you need to set it manually.