vitotai/BrewPiLess #223: 4.0 issues?

Usernametakenasdf, 2020-03-10:
Just messing around with 4.0 and finding it not saving hostname or certain settings upon reboot (pressure sensor on for example).

Usernametakenasdf, 2020-03-10:
Not saving Wifi data either.

vitotai, 2020-03-10:
I can't reproduce your issue except the ADS1115 setting.
Mine just works as expected.
The ADS1115 setting is a bug, which didn't show true value but was saved correctly.

Usernametakenasdf, 2020-03-10:
Perhaps I need to erase the Wemos and re-flash.

terragady, 2020-03-13:
Why there is a separate bin for d1pro? Is it now any different from normal wemos d1? I have flashed pro before normally as the normal d1 without issues.

vitotai, 2020-03-14:
There wasn't a device profile, or script, for D1pro. While mine worked just fine, some people just couldn't make it work.
Now that the building script is available, I see no reasons not to use it. Moreover, the dedicated script enables full 16M flash access.

vitotai, 2020-03-16:
I did find a bug that would make connection at boot up fail. Please try the new update, later.

terragady, 2020-03-27:
> 
> 
> There wasn't a device profile, or script, for D1pro. While mine worked just fine, some people just couldn't make it work.
> Now that the building script is available, I see no reasons not to use it. Moreover, the dedicated script enables full 16M flash access.

Nice, thanks!
Is there any benefit to use ESP32 now as I see it is supported?

vitotai, 2020-03-28:
1. Rotary encoder support.
2. ESP32 runs faster.
That's it so far.
The purpose to use ESP32 is BLE for TILT support and SSL. However, current built-in BLE library is too huge that it exhausts almost all memory. When a lightweight alternative, NimBLE, is available, I will try it.
SSL is a smilier story.

terragady, 2020-03-28:
I see, I don't use any encoder /button or screen, all online as the unit is hidden under the freezer. I use only ispindel so I guess there is no point changing old d1 Pro for esp32.

ollinator2000, 2020-04-08:
> Not saving Wifi data either.

Mine Wemos isn't saving Wifi Settings either. 
In addition i can't set a http remote logging target . Maybe its a layer 8 problem , but im im constantly getting 401 failed .

vitotai, 2020-04-10:
What is your browser?
What is your Flash ID, displayed in system pages.
What action got 401 response?

ollinator2000, 2020-04-13:
> What is your browser?
> What is your Flash ID, displayed in system pages.
> What action got 401 response?

I tried with Safari.

Flash Chip ID
0x164020

I tried to setup external HTTP Post Logging

vitotai, 2020-04-15:
401 response means invalid input. It might be software issue or you having wrong input.

ollinator2000, 2020-04-15:
Im running a simple python http Server which just prints all incoming HTTP Posts . 

I setup in brewpiless
Remote Log
Period 1 
Service Type Generic HTTP
Method Post
URL "IP of my Python Server":8008
Data Type application/json

Do you see any mistakes i was configuring... ?

vitotai, 2020-04-15:
Format?

Period is too short.

shambolic94, 2020-04-18:
I've also been having a problem with brewpiless periodically forgetting my wifi data (maybe once a week or so), requiring me to log in to the brewpiless AP and re-enter network name, password, and ip address.  I use chrome and flash chip ID is 0x16400b.  It seems to happen randomly as far as I can tell.  I'm using the latest 4.0 bin, but was having the problem in the last few v4.0pre versions as well.

ollinator2000, 2020-04-20:
> Format?
> 
> Period is too short.

i  missed the format field. my fault ! but i can't get brewpiless send some data . here is what i configured. 

i expanded the log period to 60 seconds and the following format string
fridge=%f&room=%r

vitotai, 2020-04-23:
> I've also been having a problem with brewpiless periodically forgetting my wifi data (maybe once a week or so), requiring me to log in to the brewpiless AP and re-enter network name, password, and [...]

I am resolving this issue.

vitotai, 2020-04-24:
> I've also been having a problem with brewpiless periodically forgetting my wifi data (maybe once a week or so), requiring me to log in to the brewpiless AP and re-enter network name, password, and [...]

I just push an update addressing this issue. It might take a while until I know it's solved, but you can try if you want.

jnr-nat, 2020-04-28:
Im new to coding. I use a Thorrak PCB with my BrewPiLess. I want to run OLED which current the THORRAK BIN doesn't support. Is there a way a can change the code that will allow v4.0 OLED BIN to use pin 0 for Heat?

vitotai, 2020-04-28:
> Im new to coding. I use a Thorrak PCB with my BrewPiLess. I want to run OLED which current the THORRAK BIN doesn't support. Is there a way a can change the code that will allow v4.0 OLED BIN to use pin 0 for Heat?

Try to study the platformio.ini file, you should know the differences between each option. It's a good start to begin in coding.

jnr-nat, 2020-04-28:
Sorry, I downloaded Micro VS Code program and the platformio extension. I can open the whole repository and see all the BIN files but unsure of how to open and edit a .bin file.

If you have a good video or blog to that I can read up it it would be awesome

Naesstrom, 2020-04-28:
> I just push an update addressing this issue. It might take a while until I know it's solved, but you can try if you want.

I'm having problems with it loosing the wifi settings to on v4, can't see any updated files since mars 26 but you posted this a couple of days ago?