vitotai/BrewPiLess #224: DS18S20 supported?

nicr4wks, 2020-03-29:
Hi

I have a bunch of DS18S20 sensors which came directly from Maxim as engineering samples so safe to assume they're genuine.

I can not get them detected in BrewPiLess 4.0 (or earlier versions) as temperature sensors, my ebay DS18B20 are all fine.

Is this sensor supported? The DS18S20 is supported by the standard OneWire library as I've tested them in several demo sketches.

Chip info as read by the library in case any relevant info.

ROM = 10 A6 E1 96 2 8 0 D3
  Chip = DS18S20
  Data = 1 3A 0 4B 46 FF FF B 10 2F  CRC=2F
  Temperature = 29.06 Celsius, 84.31 Fahrenheit

Thanks.

allthepies, 2020-03-29:
Well BPL uses the standard lib so you'd think it should work.  I see that the DS18S20 has fixed resolution, perhaps BPL is attempting to set the resolution and this causes issues ?  (Will look at code later).

lalo-uy, 2020-03-29:
Looks like you hace to define REQUIRESDS18S20MODEL In Config.h and recompile

 // support for DS18S20
#ifndef REQUIRESDS18S20MODEL
#define REQUIRESDS18S20MODEL false
#endif

allthepies, 2020-03-29:
> REQUIRESDS18S20MODEL

Yes, good spot!

nicr4wks, 2020-03-30:
> Looks like you have to define REQUIRESDS18S20MODEL In Config.h and recompile

Thanks for finding that, am I looking in the correct file as I can't find these settings?
https://github.com/vitotai/BrewPiLess/blob/master/src/Config.h

nicr4wks, 2020-03-30:
I've added the following to Config.h (inserted at line 189)

#ifndef REQUIRESDS18S20MODEL
#define REQUIRESDS18S20MODEL true
#endif

Compiled. Uploaded BrewPiLess-master\.pio\build\classic\firmware.bin to my NodeMCU.
init eeprom > refresh devices, no temp sensors listed.

First time using Platform.IO & vscode so please advise if I've done anything wrong.
Thanks.

allthepies, 2020-03-30:
Try setting  REQUIRESDS18S20MODEL=true as a directive in platformio.ini under build_flags ?

i.e. adding -DREQUIRESDS18S20MODEL=true

nicr4wks, 2020-03-30:
Modified as follows..

build_flags =
    -DOLED_LCD=true
    -DSerialDebug=true
    -DREQUIRESDS18S20MODEL=true

Still no temp sensors showing up.

nicr4wks, 2020-03-30:
Managed to find some DS18B20 sensors with a sub 2 week postage time.

I'm happy to do more testing if anyone else is interested otherwise please ignore and I'll setup the supported sensors when they arrive.

vitotai, 2020-03-31:
BrewPi, and hence BrewPiLess, does not accept temperature sensors in parasitic mode. According to Elco, the greater of BrewPi, parasitic powered sensors are not stable. I haven't dig into this issue and haven't done anything about it.

nicr4wks, 2020-03-31:
DS18S20 was definitely powered (DS18B20 sensor working alongside). The difference seems to be the sensor resolution.

DS18S20: Fixed 9-bit resolution (1/2°C). Measures temperature in 750 mS.
DS18B20: Resolution of 9 to 12 bits (12-1/16°C).

I seem to be the only one trying to use them, I'll re purpose and have more B sensors on the way.

vitotai, 2020-03-31:
> It also identifies itself differently, 0x10 for the S and 0x28 for the B.
> 

This is it. When scanning OneWire bus, only address of 0x28 are recognized.

closed 2020-03-31 by nicr4wks