vitotai/BrewPiLess #226: Graph shows wrong values

fnivert, 2020-04-17:
It seems the graph are not always showing the correct values. In the example I have a running ramp going downward. Everything looks fine until time 14:45, where the graph shows a step up for both beer temp and beer set. I also log to Brewfather and there I do not see this step. The numbers shown to the left seems to show the correct values. Is there issues with the graph?

[image]

I then made a total reload of the page, and then the step for beer temp is not there, but still a step for beer set:
[image]

I you wonders why be beer temp and fridge temp is the same, and also the fridge set and beer set is same, that is because I have set tp, ti and td to 0, since I have a special setup with a glycol chiller without reservoar.

fnivert, 2020-04-17:
I just saw that the step in plot 2 is at another time than in plot one. A mystery where the steps come from. Now when some time has passed everything looks normal again. No unexplained steps in the plots:

[image]

autofabrik, 2020-05-07:
I have the same issue almost like it replays data from earlier in the graph. Always at the same location at the end of the graph.

jangevaare, 2020-05-08:
I've noticed this one for awhile. I expect it's an issue relating to a rotating cache for all the temp data.

vitotai, 2020-05-16:
Please provide the log if possible.