vitotai/BrewPiLess #228: Resolve conflict - between having ADS1115 and SSD1306 connected to the I2C bus

Hwerow, 2020-04-19:
Resolve the conflict between OLED SSD 1306 and ADS1115 addresses 
I don't pretend to understand all the details here
https://randomnerdtutorials.com/esp32-i2c-communication-arduino-ide/ 
 but it would seem that an ESP32 can have 2 I2C bus interfaces defined so that one could be used for the OLED SSD1306 and one for the ADS1115 for pressure sensing.
Could this be done within BrewPiLess?

vitotai, 2020-04-24:
I don't think it's impossible. If I2C library isn't singleton, it's easy to create another I2C bus dedicated.
However, PINs are precious resource and should be saved whenever it's possible.

Hwerow, 2020-04-24:
Fair point about the pins. However, I think that for ESP32 a good combination would be OLED, actuator switch, with pressure monitoring and control.

vitotai, 2020-04-28:
If OLED is used, the conflict should not happen.
SSD1306 uses 0x3C(or 0x3E), while ADS1115 can be selected from 0x48, 0x49, 0x4A, and 0x4B.