vitotai/BrewPiLess #231: Change AP IP

Usernametakenasdf, 2020-05-07:
I run two BPL and two ispindels and had a wifi outage the other day meaning that my ispindels and BPL weren’t talking. 
I have attached one ispindel to the AP of one of the BPLs but as the .1.4 IP is common, I can’t do the same with my second setup. 
Is it possible to be able to change this on the web interface?

lalo-uy, 2020-05-07:
In AP mode each BPL have its own Wifi Network, so you can use the same IP
on different networks.

Usernametakenasdf, 2020-05-07:
But are they not both 192.168.1.4?

lalo-uy, 2020-05-07:
Yes, bot not in the same network,
You have to connect to one AP or the other.

Naesstrom, 2020-05-08:
Actually what you need to change is the network name, go to System on the second BPL and change the host/network name. I just added a 1 to mine like this:
[image]
Then you just connect your second iSpindel to the network named brewpiless1! _first iSpindel should be connected to network brewpiless_
If you are running more just keep changing the network name, if it's boring with just adding numbers then feel free to name them anything you want like Joe, Ben or Cthulhu. Just make sure they have different names and connect iSpindels accordingly