vitotai/BrewPiLess #232: Invalid log!

Naesstrom, 2020-05-08:
Just finished a fermentation and saved the log, was going to view it to get the iSpindel equation but when trying to open it with the offline logviewer I get the message "Invalid Log" when trying to load it.

Log is attached:[Clara.zip](https://github.com/vitotai/BrewPiLess/files/4597448/Clara.zip)
BPL version: 4.0r1

vitotai, 2020-05-16:
use Latest log viewer:
[image]

Naesstrom, 2020-05-16:
And where can I find that, latest seems to be 2.6?
[image]

vitotai, 2020-05-16:
sorry, it's moved to 
https://github.com/vitotai/BrewPiLess/tree/master/htmljs/dist/english