vitotai/BrewPiLess #233: Wifi issues 4.0

daCo11, 2020-05-08:
I recently installed BPL 4.0 on a Wemos D1 from scratch. BPL was loosing its wifi config constantly. Meanwhile I have updated to 4.0r1 and let see how this works. However during my investigations in solving my wifi problem I recognized that my D1 is having "cn" as the country wifi setting. CN stands for China and probably this value is set from default. Could there be an issue if your local wifi access hotspot is configured according to the country you're living in and hence differently to the BPL? Many thanks for your advise.

jangevaare, 2020-05-08:
Usually wifi country settings are to stay within regulations relating to spectrum and maximum gain by jurisdiction, but it's actually fairly standard globally for 2.4ghz, which ESP8266 uses. If it had an effect, it would likely only come into play when brewpiless was an AP, if it happened to select a channel that's not permitted in the US.

daCo11, 2020-05-08:
@jangevaare you're probably right. The country code might only be relevant with APs. BPL acts as a AP though. I ran a diagnostic report with my mac and that fellow reported the mismatch in country codes. Most likely it saw the BPL acting as an AP.
However I'll further examine how 4.0r1 performs. That might have solved my problem already. Thanks