vitotai/BrewPiLess #234: ESP32 onewire temperature sensor not connecting

food-always-food, 2020-05-11:
Hello, I'm having an issue where I can't connect a D18b20 to brewpiless. I have taken a look in Config.h and found that the pin for onewire connections is pin 23 for the ESP32. I've connected the sensor to that pin and I can't seem to add it under 'Device' on the web interface.

I've tried several different ones, but no matter what they don't seem to get identified by the system. 

I have also tried uploading the software with this setting commented and not: 
`#ifndef BREWPI_SENSOR_PINS
#define BREWPI_SENSOR_PINS 1
#endif`

both ways didn't work.

I'm using the little pre-soldered boards that have the resistor already in place and I have 4 of them, none seem to function.

The board I'm using is the ESP32 DOIT Devkit development board. I'm compiling and flashing using the platformio extension for Visual Studio Code from Ubuntu.

Do you know where I could be going wrong?

stefschin, 2020-05-11:
Hi, This sensors needs a Pull Up resistor, with 4.7KOhm between Data and Vcc is ok, have you this resistor?

food-always-food, 2020-05-11:
I really appreciate that response. You're right, that was it. I've wired it all up now and it all is running fine. LCD is working, Relays are firing and the temp sensor is getting readings. 

Thanks so much!

closed 2020-05-11 by food-always-food