vitotai/BrewPiLess #236: Document sensor calibration

ashgillman, 2020-05-20:
I have a DS18B20 that seems to read differently to others by a few degrees. I notice on the setup page for each probe there is a calibration field, but I'm not sure how this works.

I'd be happy to make a PR with an update to the instructions if someone is able to explain to me what this field does.
Is this the same as the `U{i:0,j:0.36}` command?

vitotai, 2020-05-24:
You can set it under "device setup" page after v4.0.

ashgillman, 2020-05-25:
Yes, but what does this do? Scale the reading? Add/Subtract?

Its only mentioned here: https://github.com/vitotai/BrewPiLess/blob/691a92b89435d3a1124f79776f9185f83488bcc3/doc/JsonCommand.md#sensor-calibration

It sounds like it just adds/subtracts from the sensor reading?

ashgillman, 2020-05-25:
Ah, this seems to explain that yes, it is added to the reading.

https://github.com/vitotai/BrewPiLess/blob/0d9f95dfac49298b7f25ba5697f68e69003a9b74/src/OneWireTempSensor.h#L41