vitotai/BrewPiLess #237: Struggling to connect to MQTT gateway

demessmaker, 2020-05-22:
Hello,
I am trying to setup remote logging to Google Cloud via a MQTT gateway installed on a Raspberry PI next to my BPL.

Has anybody had success with a similar setup?
Can you provide any insight as to the proper configuration?

How should the base path be defined ?

[image]

Thanks

vitotai, 2020-05-24:
MQTT port is default to 1883. You might be able to change it, though.
What MQTT broker do you use? I don't have much resources, and only cloudmqtt.com and ubidots.com were tested. (However, I ran out of all my ubidots.com credit now.)

demessmaker, 2020-05-24:
Thanks for the quick answer.

I am using the gateway provided in one of the Google Cloud Platform tutorials.
https://github.com/GoogleCloudPlatform/community/tree/master/tutorials/cloud-iot-gateways-rpi

Using the provided clients, I can get the gateway to work without any problems.
I then try to get BPL to use the gateway without success.

From what I can tell, the gateway is built on Paho MQTT.
https://github.com/GoogleCloudPlatform/community/blob/master/tutorials/cloud-iot-gateways-rpi/cloudiot_mqtt_gateway.py

Thanks again

vitotai, 2020-05-26:
I don't know about the broker.
An ethernet sniffer might be a good tool to debug issues like this.

demessmaker, 2020-06-01:
Hi Vitotai,

I did manage to determine the issue was with the brocker and not with BPL using wireshark as per your suggestion.
Thanks and keep up the good work

closed 2020-06-01 by demessmaker