vitotai/BrewPiLess #238: 4.0r1 WiFi Connection problem

Hwerow, 2020-05-24:
The sequence is this

vitotai, 2020-05-24:
192.168.4.1 is the IP for AP mode. If you connect to the network BPL creates, default to 'brewpiless', then 192.168.4.1 is always the IP of it.
If STA mode is enabled, AP+STA or STA, BPL will get another IP from the network it connects to, which is the one shown on the page. If you connect to the network which BPL connects to, you should be able to access BPL via the IP address.

samysilva, 2020-06-30:
I am using version 4.0r1; after connecting to the AP; I configure the Wifi network and parameterize it to work in Station mode (to hide the AP network). The device remains in operation for some time; but some time later he loses the Wifi connection; where I need to reconnect to the AP and reconfigure the network.

vitotai, 2020-07-01:
You might try latest 4.0r2.trial version.

Hwerow, 2020-07-01:
Thanks, I will give it a go once the current brew is done.

vitotai, 2020-07-02:
Sorry. I patched the wrong branch. Please check it one day later.

Juan-gimenez-globant, 2020-07-12:
I tried with master and v4.0pre branches, I'm facing the same issues with both...