vitotai/BrewPiLess #239: [Question] How can I improve the regulation?

daCo11, 2020-05-25:
This is not an issue and more of a question in how to improve the work of the controller (if at all). Pls note the following screenshot. Is the controller doing a good job here or do I need to tweak some parameters? In case yes which one do you recommend? Beer Profile - Beer temperature set is for 19°C and the overshoot goes up to 19.2°C and the undershoot down to 18.7°C approx. Many thanks for your hints.

[image]

vitotai, 2020-05-26:
People have struggled to get a very good result when using heating and cooling devices at the same time since the beginning of BrewPi. Yes, the original BrewPi.
Some information I learned by googling:
- the heating is over powerful, which causes the rising of temperature is too fast.
- minimum "times" should be set correctly. For example, minHeatTime/minHeatIdleTime should usually be a very small value, like 10. To prevent damaging compressor, minCoolTime/minCoolTime are usually a larger value, like 300 or 5 minutes. There is no such issue for heating.
- You can play with idleRangeH/idleRangeL/heatTargetH/coolTargetL and find a good pair of values.

daCo11, 2020-05-27:
I just passed a phase with more than 10 hours where the BPL run in idle status. Wow, I might have found THE setting :-) Probably not (yet) because above screenshot was taken during the main fermentation phase where there is of course more activity inside. That main activity is now over and furthermore I have more or less stable room temperatures which makes it easy for the BPL to stick to the constant temperature. The fridge itself is not cooling or heating that much off. Let's see how it develops when there are changes in the setting of the beer temperature coming. Many thanks.

daCo11, 2020-06-01:
Meanwhile the fermentation calmed down and the BPL needs hardly to adjust the temperatures. However if there is a need for cooling it will undershoot too hard so that a heating phase follows just after a couple of minutes later. I can't understand why it's cooling for three or four full minutes even I adjusted the minCoolTime down to 60s. The calculated time for heating seems just about right. Is there a way to soften the time of cooling but not influencing the time of heating?

vitotai, 2020-06-02:
If you are using a fridge/freezer, minCoolTime should be longer to save the compressor.

There is an estimation algorithm that is calculated in each loop to determine a better regulation. Frankly speaking, I am not familiar with this part.

minCoolTime and minHeatTime are separated parameters that control minimum cooling and heating time. Changing minCoolTime should not affect heating.

I would suggest coolTargetL to a smaller value to aim a "higher" cooling target.