vitotai/BrewPiLess #241: Issue making a http POST request to BrewersFriend API

Connor-Ecocentric, 2020-05-28:
Hi i am having issues with getting the generic http POST to work. i have attempted to make the request via postman and it works fine, when attempting to POST via brewpiless i have not managed to get it to work. Is there anywhere i am able to view logs to see what is wrong with the request? please see below my settings in brewpiless.

I have also noticed a popup "https not supported" i am currently running v3.6 on a wemos D1 mini.

[image]

allthepies, 2020-05-28:
Brewpiless doesn't support secure http (https).

Connor-Ecocentric, 2020-05-28:
so how is there a workaround for this? as brewpiless does have an option for brewersfriend so i assumed it is supported? see below: 
[image]

vitotai, 2020-05-30:
I don't have premium account of Brewersfriend.com. Therefore, the support of Brewersfriend.com is implemented without verifying. I will try my best to check this.