vitotai/BrewPiLess #242: Strange behaviour - Mode "R" and "Ü"?

daCo11, 2020-05-28:
Pls have a look at the following screenshot:
[image]
What does that mode "R" at around 07:30 mean and why the heck get my "beerSet" to "-17.7" all of a sudden? It should be still at "19" and the log tells me exactly so:
[image]
Any ideas?

lalo-uy, 2020-05-28:
R means reset, most likely power failure.

daCo11, 2020-05-29:
So funny. This morning we're back on track with "beerSet" in the graph (as said the log never told anything else) to "19":
[image]

vitotai, 2020-05-30:
Please provide the log, if possible. Something wrong with the logging or graph display.

daCo11, 2020-05-30:
> Please provide the log, if possible. Something wrong with the logging or graph display.
Sure I'll do. The screenshots are from a running batch. Would I need to stop logging for getting the file or can I just download it from the BPL? Futhermore I'm struggling to post that log file ...

daCo11, 2020-06-01:
@vitotai I finally figured it out how to publish a logfile. Here the link to that log on my github repository: [Batch006](https://github.com/daCo11/daCo11.github.io/blob/master/BPLlog/Batch006)

The strange behaviour happend around the 29. May. I tried to open that logfile myself with one of your BPLLogViewer.htm files but I keep getting an "invalid log!" error message. What am I doing wrong?

vitotai, 2020-06-02:
Please use the viewers in htmljs/dist folder. 
I am busy these day and will check this issue later when I have time.

daCo11, 2020-06-02:
> Please use the viewers in htmljs/dist folder.
> I am busy these day and will check this issue later when I have time.

@vitotai I got it, many thanks. There is no hurry for looking into that. As mentioned the controller continued to work as wanted. It was just the graph displaying something odd.