vitotai/BrewPiLess #243: Remote log going back to default

terragady, 2020-05-30:
Hi, when I go to Remote Log, and set stuff for for example BrewFather, press update and refresh the page, it shows me the default Remote Log page (like Generic HTTP and empty inputs). I can see on my brewfather that some of the log is sent. So my question is is this a bug or it is set to have multiple remote logs and the saved log settings are saved somewhere else? If it is the case how I can stop remote logging?

bhackett14, 2020-09-15:
Having the same issue. Worked well for my last batch but all of the sudden it deleted the saved Brewfather data. Now when I try to save it, it deletes it a reverts back to Generic HTTP.

bhackett14, 2020-09-17:
Never mind. It did not show Brewfather credentials but it did save. The Brewfather app is now updating again.