vitotai/BrewPiLess #246: 2nd iSpindel for display only?

Scales82, 2020-06-22:
Hey @vitotai 
I realise that at the moment BPL only allows the connection of one iSpindel for the use of temp control and display etc.
I was wondering, would it be difficult to add functionality to allow for a 2nd (or even 3rd I guess) iSpindel for only displaying gravity and temp? If the data could also be forwarded onto Brewfather etc, that would be even better!

I have a fridge that holds two fermenters, and I would much prefer to have the data all displayed in the one location, rather than having to look elsewhere.

vitotai, 2020-06-24:
It's not difficult, but say it might take one hour for coding and one week for verifying...

Scales82, 2020-06-24:
Happy to help where I can!