vitotai/BrewPiLess #248: Negative temperature

samysilva, 2020-07-08:
Negative temperature

Is it possible to modify the code so that negative values ​​are allowed?

I have a fermentation profile; maturation; clarification; and special ramp (high gravity) that needs to consider negative temperature.

However, I cannot reach the internal temperature of the fermenter by 1 degree; because the algorithm parameterizes the lowest dynamic temperature of the "fridge set" by 1 degree. What does not create a temperature delta with enough difference to decrease the temperature inside the fermenter.

The freezer turns on at around 2 degrees and turns off at around 1; and turns on again at 2. As the fridge set is never less than 1 degree; is causing unnecessary energy consumption by running the engines all the time.

It would be possible to change the code so that the algorithms consider negative values. And also; allowing me to set the freezer to negative temperatures when I need it?

I am getting around using a common temperature controller; but it is not as practical as brewpiless; because it doesn't have all these ramp features; etc

Thank you

lalo-uy, 2020-07-08:
The minTempSet can be modified with command.htm, check the wili

samysilva, 2020-07-08:
I didn't find this file.

I'm studying the code.

I see that the variables used to store the temperature are uint16_t "UNSIGNED SHORT INT 16" (which allows values from 0 to 65535).

Initially I imagined that it would be necessary to change all types of these variables to int16_t "SIGNED" (in order to allow values -32768 .. 32767).

Studying the code a little more, I found in "TempControl.cpp" the function that defines the initialization variables.

I performed a test changing the "tempSettingMin" to -10; compiled; and I'm running tests.

For my surprise; this change alone was enough for the algorithms to consider negative values (although almost the entire implementation is UNSIGNED); how it worked; I didn't continue to study the code.

I will continue testing; if it is just that; it will be great, because I will be able to use the freezers for other purposes including.

lalo-uy, 2020-07-08:
My fault, the url is:

http://brewpiless.local/testcmd.htm
Look here : https://github.com/vitotai/BrewPiLess/wiki/JSON-command

   j{"tempSetMin":-5.0}

samysilva, 2020-07-08:
With the limit parameterization at -10; it became much more efficient for me.

When I set the temperature to 1 degree, for example; The fridge's dynamic setpoint is set to - 0.5; -0.1, etc; This creates a suitable delta so that the internal and external temperature difference can reach the desired 1 degree.

Another thing I noticed is that my freezers are now working much less; Perhaps it would be beneficial to leave the value by default at -5 (for example). It will probably help other users who don't know how to interact with programming or api.

Talk soon