vitotai/BrewPiLess #249: Device Setup not recognizing

bhackett14, 2020-07-14:
Double and triple checked my wiring but the relays and temp sensors aren't being recognized. I don't have anything on D0 but it refreshes with it and even when I unplugged D7 it still populates
[image]

kall3, 2020-07-14:
I am having the same issue. Running version 4.0 classic on a ESP8266 (d1 mini)

kall3, 2020-07-14:
> I am having the same issue. Running version 4.0 classic on a ESP8266 (d1 mini)

Don't mind me, I had soldered my 6,3 mm stereo jack faulty, hence no onewire initialization

bhackett14, 2020-07-14:
@kall3 were you seeing the same "detected" devices? D0, External Sensor stating N/A, etc.? If so maybe my wiring is also faulty to my onewire setup.

kall3, 2020-07-15:
@bhackett14 yes that's right. I had soldered the female stereo 6,3 mm jack faulty so that I was hooking up the live wire from the DS18B20 to GND in the connection socket. Once I got that right everything is working and my BPL has been controlling the fridge and a PTC-element since yesterday.

bhackett14, 2020-07-20:
*User error: Platform IO was installing all binaries instead of defualt v4.0. To make it easier for myself I just downloaded the binary for v4.0 and used the flashing tool to download it to the ESP8266*

closed 2020-07-20 by bhackett14