vitotai/BrewPiLess #250: Setting the Fan stop time

samysilva, 2020-07-15:
I use the "**Chamber Fan**" function to activate a fan inside my freezers to circulate the air; when heating; or cooling is triggered.

At first; the function works in parallel with the "Chamber Heater" or "Chamber Cooler" functions.

When the heating is turned off; the fan is turned off immediately.

I want to modify the code; adding a condition that for the fan to continue running for 2 minutes; after heating stops.

Implementation is simple; I just need to know where is the routine that stops the fan; to change the method; calling another method that does is performed after the time I wish.

This will allow my system to improve the heat exchange; because after the heating stop; the heater remains hot; generating a pocket of heat that interferes a lot in the balance. This will also help on the cold side; as the freezer continues with the frozen plates after stopping.

samysilva, 2020-07-15:
Sharing: I was able to identify the actuators; I created a new class that considers a timeout value for the execution of the stop; and changed the type of the fan class to the new one. It's working correctly. If this is useful there is someone, I send the how to do it.

kall3, 2020-07-15:
@samysilva are you using a SSR for the fan just like with the heating and cooling? or are you using a low voltage computer fan?

samysilva, 2020-07-15:
I am using 3 common 20A relays; I use SSR only in the boiling pot because of PWM. My fan is also 127v. With the new logic; the fan is staying on for 3 minutes after the heater or freezer turns off.

kall3, 2020-07-15:
That's a nice one, please share how this is done code vise

samysilva, 2020-07-15:
Attached is the new actuator; and the files that make use of it properly updated. Work like a charm.

[ActuatorDelayedOff.zip](https://github.com/vitotai/BrewPiLess/files/4925901/ActuatorDelayedOff.zip)

1maddi, 2020-07-28:
I‘ve tried this too but without effort.
It would be great if you can provide me your code!

terragady, 2020-08-10:
yeah some more FAN options would be nice to have, like not only when it is cooling or heating, but also when for example there is no heating or cooling for 1 hour the fan should do a 5 min round for example for the sake of mixing the air

acremonezi, 2020-12-23:
I would also like to see this code.

I am planing to use two fans into the chamber, in order to make a continuous air flow circulation.
One fan each corner, Flowing from Up to Down and Down to up, to promote a better circulation, instead of just waiting the natural air flow convection due to the temperature difference.

**To implement this, I am thinking of a behavior like this:**

- Two Fans connected into the same micro-controller Pin attached to a Transistor as a Switch to provide 12v to the fans. (I can share circuit diagram later on)
- The 2 Fans always ON (Never stop) but with a Duty Cycle PWM of 30%. (Can be more or can be less, it is just to sustain the air flow circulation cited above without making too much noise)
- When Heating or Cooling is ON, change the PWM to 90%
- When Heating or Cooling is switched OFF, keep PWM to 90% for 1 minute more. (Just to stabilize)
- If Chamber Door is Open, switch OFF the Fans.

What do you think guys?