vitotai/BrewPiLess #251: ESP32 Failing on upload

bhackett14, 2020-07-15:
Uploading of ESP32 fails each time I try and upload but everything else works fine. I have tried even specifying the COM port. The first time I uploaded I didn't notice this and everything seemed to be working fine. I then wasn't able to detect devices correctly. I am assuming that is related?

Here is the log:

`> Executing task in folder BrewPiLess: C:\Users\bchacket\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing esp32 (platform: espressif32@~1.9.0; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.9.0 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 2.10002.190628 (1.0.2)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 6.10.0
|-- <ESP8266_SSD1306> 4.0.0
|   |-- <Wire> 1.0.1
|-- <FS> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <AsyncTCP> 1.1.0
|-- <ESP Async WebServer> 1.2.2
|   |-- <AsyncTCP> 1.1.0
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.10.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <OneWire> 2.3.2
|-- <HTTPClient> 1.2
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <WiFiClientSecure> 1.0
|   |-- <WiFi> 1.0
|-- <Wire> 1.0.1
|-- <Update> 1.0
|-- <WebServer> 1.0
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <EEPROM> 1.0
|-- <AsyncMqttClient> 0.8.2
|   |-- <AsyncTCP> 1.1.0
|-- <Adafruit ADS1X15> 1.0.1
|   |-- <Wire> 1.0.1
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
Building in release mode
Retrieving maximum program size .pio\build\esp32\firmware.elf
Checking size .pio\build\esp32\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  15.4% (used 50304 bytes from 327680 bytes)
Flash: [========= ]  89.3% (used 1171022 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa      
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: COM6
Uploading .pio\build\esp32\firmware.bin
esptool.py v2.6
Serial port COM6
Connecting....

A fatal error occurred: Timed out waiting for packet header
*** [upload] Error 2
==================================================================== [FAILED] Took 8.40 seconds ====================================================================`

bhackett14, 2020-07-15:
Do I have an under-powered ESP8266. This is what I have and says 80Mhz CPU

https://www.amazon.com/gp/product/B01IK9GEQG/ref=ppx_yo_dt_b_asin_title_o08_s01?ie=UTF8&psc=1

But the log says it is trying to upload to hardware with 240Mhz

`HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash`

vitotai, 2020-07-17:
usually it's USB to uart driver issue.

bhackett14, 2020-07-20:
@vitotai if I am using a ESP8266 such as the one I am using from Makerfocus should the ESP32 always fail since it is not a ESP32?

bhackett14, 2020-07-20:
*User error: Platform IO was installing all binaries instead of defualt v4.0*

closed 2020-07-20 by bhackett14