vitotai/BrewPiLess #252: Remote Log with Generic HTTP will not save

inthesticks, 2020-07-16:
I'm trying to report HTTP from BPL to Brewfather so that I can get pressure to report but once I click update the changes are not saved and just revert back to Brewfather service type after clicking the update radio button

inthesticks, 2020-07-16:
It seems as soon as i enter the BF URL http://log.brewfather.net/brewpiless?id=xxxxxxxxxxx and click update it then reverts to the service type Brewfather

vitotai, 2020-07-17:
You are right. Everything is generic http logging. The easiest way for you is adding the pressure vale to the preset of brewfather.

inthesticks, 2020-07-17:
> You are right. Everything is generic http logging. The easiest way for you is adding the pressure vale to the preset of brewfather.

Do you know why I can't save the new log after changing to HTTP, I've even tried on older firmware on another D1 and still won't save

vitotai, 2020-07-17:
If you see "Done" or "OK" response, it might be file system issue. Otherwise, something wrong with the setting values.
By the way, pressure value is already included in brew father's preset.

inthesticks, 2020-07-17:
> If you see "Done" or "OK" response, it might be file system issue. Otherwise, something wrong with the setting values.
> By the way, pressure value is already included in brew father's preset.

I do receive "Done" but the service changes from HTTP back to Brewfather only when the Brewfather URL is added.

Pressure value is included in the service Brewfather? ive looked on Brewfather and dont see any settings to turn on as it doesnt appear on the readings

Cheers

vitotai, 2020-07-17:
Are you sure that Brewfather support pressure? What label do they use for pressure?

inthesticks, 2020-07-21:
I have recently found out that Brewfather currently doesn't support this feature but i have created a request on Gitlab to do so.
once done it can be tested