vitotai/BrewPiLess #256: Reset option

Scales82, 2020-08-01:
Hi Vitotai,
Love BPL. its the best.
Is there a command that can be issued by JSON or similar to reset the BPL over the gui/remotely. I am having some minor problems and having to power cycle the device is becoming problematic (especially remotely).
Cheers

vitotai, 2020-08-02:
There was. It's easy to add it back.
What's your issue?
IMO, SPIFFS seems to be a weak point of the system, in addition to WiFi. I would suggest formatting SPIFFS if the system is unstable.
The url to format SPIFFS is
http://{ipofthecontroller}/format-spiffs

Scales82, 2020-08-02:
Cool thanks.
The issue im having is every so often all 3 of my temp probes disappear entirely from the system. conducting a hard-reset is only way for them to be re-initialised and have it all come back to normal. As there is no reset feature I have to manually go to the fridge to do this.

vitotai, 2020-08-03:
I had similar issues before removing DIN connectors.

lalo-uy, 2020-08-03:
In my experience that happens for supply interference.
Power the sensor with 5v.
Add 100nF between 5v and Gnd as close to the sensor as possible.
Lower the data pull up to 3v to 1k ohms.

terragady, 2020-08-10:
I have also sometimes problem with sensors, I have 2 of them and sometimes they spike abnormally on the graph and they show then the same temperature and If I check on devices only one is showing temperature and the other one not. Have to power cycle, my cables are around 1m.

Scales82, 2020-09-08:
@lalo-uy Thanks. Is the cap absolutely necessary? could I get away with just the 1k swap with 5v instead of 3v3?

lalo-uy, 2020-09-08:
Do not use 5V in the data pullup line, that may destroy the input pin. ESP
do not allow 5v in the pins.

terragady, 2020-09-08:
I did use many times 5V and it was alright, with ESP, Arduino and Raspberry. No probs. Here though I am using 3v3 as it is super easy to just solder resistor to pins

lalo-uy, 2020-09-08:
That your specific device tolerate that not means it "is all right".
Manufacturer specs say ESP and Raspberry Pi GPIO should not be exposed to 5V

vitotai, 2020-09-08:
ESP8266 is said, by the manufacturer, to be 5V tolerant.

lalo-uy, 2020-09-08:
Sorry but no. Look here.

Scales82, 2020-09-08:
> Do not use 5V in the data pullup line, that may destroy the input pin. ESP do not allow 5v in the pins. El lun., 7 sept. 2020 a las 23:25, Scales82 (<notifications@github.com>) escribió:

Thats ok, I wasn't doing that anyway. I have re-designed and put 5v supply to the probes and left the signal line at 3v3. I was wondering mainly about the change in resistor and why you suggest 0.1uF but some designers have also put a 1uF on both the 5v and signal lines.

lalo-uy, 2020-09-08:
1uF in data line is bad, will distort the pulses shape.
As mutch as you stabilize the power supply the better.
100nF or 1uF are good as long as they are paper caps. Electrolytics ones
tho not respond fast enough.