vitotai/BrewPiLess #257: Loosing sensors

kall3, 2020-08-06:
I am running BPL 4 "new ui" on a ESP8266 chip (D1 mini), it boots fine and takes commands but after 10-15 min of controlling the fridge/heater it looses connection to the temp sensors and goes in to idle... Any ideas?

vitotai, 2020-08-07:
https://github.com/vitotai/BrewPiLess/issues/256#issuecomment-667946320

kall3, 2020-08-08:
The issue seemed indeed to be long cables from the sensors to the control box. I changed the pickup between signal and power to 1kohm instead of 4,7 kohm and I powered it from the PSU with 5V instead of the 3,3V from the D1 mini.

lalo-uy, 2020-08-09:
Power the sensors with 5V, but not the pullup. The R must pull to 3.3v since that is the max alowed by the D1 pins.

Scales82, 2020-09-08:
@kall3 @lalo-uy  What does the 1k resistor change?

lalo-uy, 2020-09-08:
A lower pullup produce faster edges in the data bits and more tolerance to
noise.

Pinkimark, 2022-12-12:
I have found the same issue, however if you change  "#define FridgeSensorFallBack true" to false on line 591 of config.h and then re-upload your software this issue stops. I believe this is the setting the lets the system ignore the fridge temperature for if you are using a glycol chiller unit. - obviously if you are using a glycol chiller then this may well not be a solution for you!