vitotai/BrewPiLess #258: Large log file size prevents graph populating

Usernametakenasdf, 2020-08-07:
I'm having an issue where after a week or so, the graph stops populating.
Speaking to others suggested it was a log file size issue and as soon as I deleted my log and created a new one, the graph populated again straight away.
[image]

allthepies, 2020-08-07:
I commented on the Facebook group thread about this issue;

When I had a similar problem I found that the logfile size was getting to the point where the UI Javascript was timing out accessing the logfile (could see this in chrome developer tools).  I no longer use logging due to this.

I did wonder if a clever algorithm could be used to reduce the logfile size when being accessed for UI purposes but didn't explore further.

vitotai, 2020-08-08:
I knew it would happen. I had this issue from the very beginning when testing HTML/JS files of 100k+ size. 
Formatting SPIFFS every few brews to keep SPIFFS in good shape would help. However, LittleFS is the way to go.

https://github.com/esp8266/Arduino/pull/7263

lbussy, 2020-08-08:
I moved to LittleFS for Brew Bubbles and the documentation to change the words took longer than the code changes.  Unless you are doing something Uber-esoteric, it should be a handful of changes.