vitotai/BrewPiLess #259: No beer sensor assigned but beer temp is shown

meirion, 2020-08-16:
Hi, new user to brewpiless here from Fermentrack.

I have a strange issue - I have 3 temp sensors, but one does not show up in the device list so I assume it is dead.

The other two sensors are assigned to chamber and room temp.
However a beer temperature is displayed on the graph - where is this temperature coming from? It is very close to the chamber temperature.

[image]
[image]

meirion, 2020-08-17:
I fixed my third sensor - wire had become disconnected. I am still confused how I had three temp readings with only 2 detected sensors however.

vitotai, 2020-08-18:
The picture I got in the mail shows that "Beer" sensor is set. It is a feature called FallBackSensor for Glycol temperature control. When Chamber sensor is missing, the controller will read Chamber temperature from Beer Sensor. Please check Glycol control related document for more detail.

If only Chamber sensor is set as you stated, I don't know why. It's a mystery to me.

meirion, 2020-08-18:
Yes, I saw in my screenshot I uploaded it did have the beer sensor configured so I must have made a mistake - but your explanation about the FallBackSensor sounds like what I must have been seeing.

Thanks for pointing that out :)

The good news is I got everything working now.

[image]

Flashing the node MCU with `blank4mb.bin` and then flashing `BrewPiLess.v4.0r2.trial.4m2m.bin` sorted out my problems.
Thank you, now my keezer is up and running again and I'll soon have cold beer!