vitotai/BrewPiLess #262: Delay to show real temperature

jaimeagaviria, 2020-08-29:
Hello, I am testing this wonderful project, I hope I can collaborate by coding a little =).

I have connected 2 sensors, one for the beer and one for the chamber, when I am on the setup.htm page I see that my sensors are listed correctly with the correct temperature, however on the display and in the web graphic I do not see that temperature reflected It's like I have a big delay, is this so? or should I check something in my settings?

The images were taken seconds apart.

[image]

And

[image]

You can see a big difference in the sensor value and what is shown in the web graph or on the display.

Thank you

vitotai, 2020-08-30:
The temperature readings in device setup page are got when "listed". They are not updated after then.

jaimeagaviria, 2020-08-31:
Thanks for the answer, the way that I used to compare these two temperatures (setup page and display) was by continuously clicking on the button that lists the devices, with this I get an updated list of the devices and their current temperature, in this way I get a temperature aligned with reality after applying cold or heat to simulate temperature changes, while the display took much longer to show a real temperature, it took a few minutes for the data shown on the display to be the real temperature.

vitotai, 2020-09-01:
Beer and Fridge temperature readings are "filtered".  You can change the filter parameters by JSON commands:
https://github.com/vitotai/BrewPiLess/wiki/JSON-command

I am not familiar with those values, though.

According to my understanding, the values on the device setup list should be "raw" value, converted directly from the sensors. I might be wrong or there might be some issues. Since it doesn't bother me, I would do nothing with it. You might try other program or test code to verify the hardware, and then try tracing code of BrewPi if you want to.