vitotai/BrewPiLess #263: FEATURE REQUEST - Humidity Control

Bananamannn, 2020-09-07:
Would it be possible to add Humidity control to the project as well? Could be done using something like a DHT22 sensor and using 2 more pins to control another pair of relays.

vitotai, 2020-09-08:
I had humidity issue in my fermentation chamber, a converted fridge. This sound to be a good feature. I will put it on my list, in the front.

pedromargarido, 2020-09-08:
That would be a cool feature, but what kind of device can be used to reduce the humidity?

lbussy, 2020-09-08:
Actively reducing humidity requires cooling or heating.  That or an even more elaborate ducting system to remove and dry the air without impacting the chamber temp.  It seems to me that active dehumidification is at odds with the purpose of this controller.  The only reason you would want to "control" humidity is if you wanted a certain level.  In our cases, there is no such thing as "too dry."

The simplest way may be best.  I use [one of these](https://www.amazon.com/Improved-Eva-dry-500-Renewable-Dehumidifier/dp/B00BD0FN8A/) with excellent results.  If you need more than that, you may be looking at air leaks that need to be closed up, a (horror!) beer leak, or maybe you just open it up too much to look at the fermentation in which case a camera or a [bubble counter](https://www.brewbubbles.com) might serve you well.

vitotai, 2020-09-08:
I use something like this:
https://www.amazon.com/Pro-Breeze-Electric-Dehumidifier-Portable/dp/B01DC5PPWM
or this
https://www.amazon.com/Schmitt-Dehumidifier-Water-Ultra-Quiet/dp/B083Y74YVV

Mine isn't either one, but the dimension is very similar. I guess they are similar.

It does generate heat, so I have to use a timer to turn it off every hour, otherwise, my fridge struggled to lower temperature when it's down to 10 degree C. 

I am living in a place with humidity regularly over 60%. It's usually 70%~80%, or 100% when raining, which happens from time to time. I've tried the passive way like EVA-dry without luck. This mini dehumidifier does the job for me. BTW, it doesn't work inside my keeper, though.

lbussy, 2020-09-08:
Those do introduce heat you will find.  I still wonder though, what value does a controller add to this?  Possibly reporting RH would be good, but what value does a control loop add?

Bananamannn, 2020-09-08:
So I've been using Brewpiless for a while and I really love it @vitotai. I don't actually need humidity control to keep my beer at a nice stable temperature, Brewpiless does its job very well.
I want to start doing some other fermenting/preserving (namely salami) that DOES require humidity control.
My thought was, why build a separate chamber and controller for humidity when I already have one that does temperature and could do humidity as well. Brewpiless just works so well and the interface is just so good I would really like to keep using it.

My other option was to try building my own program to do humidity and temperature but with my crappy coding skills it wouldn't be anywhere near as good or as functional.

quimnut, 2020-09-08:
There is also https://github.com/matsstaff/stc1000p/blob/master/rh/README.md

helgikraftman, 2020-09-18:
Hello! I am a regular user of your projects (brew maniac + distilling and brewpiless). Thanks for your development. I would like to join those asking to add humidity control functionality. Because the refrigerator is used not only for fermentation of beer, but also as a chamber for ripening cheeses.

vitotai, 2020-09-27:
Since I am living in a very humid place, condensation water on my fermenter is an issue. I've seen mold when fermenting Belgian beer at 24C, which is scaring. Keeping the dehumidifier running all the time is simple, but the heat generated hinders cooling. My current solution is a time-controlled switch, which works well. 

Therefore, my rough idea is keeping RH lower than a setting value, like 65%, to prevent mold. In the future and if possible, there might be an option for mold-prevention, which set the RH value automatically based on temperature.
https://energyhandyman.com/knowledge-library/mold-chart-for-temperature-and-humidity-monitors/

Cheese making seems to be a different story. If the requirement is well described, I might be able to consider designing it as well.

The first step will be humidity logging and display, which might takes more time than controlling.

Bananamannn, 2020-09-28:
In regards to the cheese making. It really just involves keeping a stable humidity and temperature, same as salami, aged meats, etc. So turning either a humidifier OR dehumidifier on/off I would think.

In my area I have very low humidity all year round, but the summers get up to 40C with the winters getting down to 0C. My setup would only be needing a humidifier, unlike yours @vitotai  which needs a dehumidifier.  Could this be done by simply inverting the switching logic? Or adding a second output (one for humidity up, another for humidity down)?

helgikraftman, 2020-09-28:
Thanks for reply, Vito!

> The first step will be humidity logging and display, which might takes more time than controlling.
There are problems when using a temperature (DS18b20) and humidity sensor (DHT 11/22) on one pin, as I know. Since the number of pins on the board is limited, it may make sense to use a psichometric measurement method (the difference between the temperature "dry" and "wet" thermometer).

> Since I am living in a very humid place, condensation water on my fermenter is an issue. I've seen mold when fermenting Belgian beer at 24C, which is scaring. Keeping the dehumidifier running all [...]

If you have high humidity in a closed chamber, then maybe it makes sense to use recyclable moisture absorbers - rice, salt, ethanol, silicagel cat litter?

> Cheese making seems to be a different story. If the requirement is well described, I might be able to consider designing it as well.
Just need on | off relay switch witch turn on | off with setting parameters of the upper and lower boundaries of humidity. And humidity logging ofcourse.

> The first step will be humidity logging and display, which might takes more time than controlling.
I will look forward to it.  Thanks for your work!

Bananamannn, 2020-09-29:
Another thought I just had was that since we are running out of pins on the Di Mini, we could always add these extra features to pins on a PCF8574 module since this is already supported.
I'm not sure how well a DHT style sensor would work over I2C but the buttons defiantly would since they are already supported. 

@helgikraftman - I like the idea of using the different temperature values to calculate the humidity.  I have some reading up on the topic to do I think.

PRyC, 2020-10-27:
Maybe better add support to BMP/BME 80/180/280? More accurate than DHT11/22. In addition to humidity, there is temperature and even pressure (BME). And is I2C interface.

vitotai, 2020-12-08:
> Would it be possible to add Humidity control to the project as well? Could be done using something like a DHT22 sensor and using 2 more pins to control another pair of relays.

Guess what, I did it.
Humidity chart is available, but controlling is not yet verified/tested. You might want to try it, it is now on the master branch.

> Maybe better add support to BMP/BME 80/180/280? More accurate than DHT11/22. In addition to humidity (BME), there is temperature and even pressure (BMP and BME). And is I2C interface.

I see your point. But, I happen to have a AM2301 on hand.

PRyC, 2020-12-08:
> > Maybe better add support to BMP/BME 80/180/280? More accurate than DHT11/22. In addition to humidity (BME), there is temperature and even pressure (BMP and BME). And is I2C interface.
> 
> I see your point. But, I happen to have a AM2301 on hand.

If necessary, I can send You BM*280 :-)

Bananamannn, 2020-12-08:
> Guess what, I did it.
> Humidity chart is available, but controlling is not yet verified/tested. You might want to try it, it is now on the master branch.

Fantastic! I'll really have to finish the new circuit board I'm making for it now. Can't wait to give it a go!!

Bananamannn, 2020-12-08:
> > > Maybe better add support to BMP/BME 80/180/280? More accurate than DHT11/22. In addition to humidity (BME), there is temperature and even pressure (BMP and BME). And is I2C interface.
> > 
> > 
> > I see your point. But, I happen to have a AM2301 on hand.
> 
> If necessary, I can send You BM*280 :-)

That is a very clever idea. You can easily retro fit something like that too existing units.

mtornblad, 2021-03-04:
Seems like some changes related to this made building of HTMLJS break. Needed to add this into the language.json files to be able to build it...

    "chart_rh":"!TODO!",
    "status_humidity":"!TODO!",
    "label_humidity":"!TODO!",
    "dht_temp_sensor":"!TODO!",
    "setup_sensor_type":"!TODO!",
    "dht_sensor_family":"!TODO!",
    "setup_humidifier":"!TODO!",
    "setup_dehumidifier":"!TODO!",
    "log_brewersfriend_stream_apikey":"!TODO!",

Bananamannn, 2021-10-11:
1. Trying to add humidifier and dehumidifier control pins too ESP8266 D1 Mini board that I run. Would the below changes too the "Config.h" file work (Lines 432 to 448)?

2. Also how do I enable the BME280 sensor instead of the DHT22? I believe it is supported.

3. Also when I try to compile it a get a heap of errors and it fails to compile (log attached). I don't think it is related too the HTMLJS compile issue. Any ideas anyone?
 
```
#elif BOARD == BrewShield
// default
#define oneWirePin NODEMCU_PIN_D6  // If oneWirePin is specified, beerSensorPin and fridgeSensorPin are ignored
#define coolingPin NODEMCU_PIN_D5
#define heatingPin NODEMCU_PIN_D7
#if AUTO_CAP
#define doorPin NODEMCU_PIN_A0
#define BuzzPin NODEMCU_PIN_D0
#define humidifier NODEMCU_PIN_D10
#define dehumidifier NODEMCU_PIN_D8
#else
#define doorPin    NODEMCU_PIN_D4
#define BuzzPin NODEMCU_PIN_D0
#endif

#define UpButtonPin NODEMCU_PIN_D3
#define DownButtonPin NODEMCU_PIN_D4
```
[Build Log.txt](https://github.com/vitotai/BrewPiLess/files/7318668/Build.Log.txt)

Bananamannn, 2021-10-20:
Thanks for the info on the humidity enable feature. But when I try to build it I get all the errors in my previous comment build log, do you know why?

Also when I was fiddling with the pin definitions I was trying too see if I could use a D1 Mini to run 4 different relays.

StuartKing1, 2023-12-12:
I have connected a DHT22 to PIN D8 (GPIO15) as this is what is shown in the device list of BPL V4.2, The ESP8266 does not boot. 

The spec says that if this PIN D8 is pulled HIGH it won't boot, how do I configure the software to use an alternative PIN or get around this boot issue?

Can I use D1 or D2 GPIO5 or 4?

Thanks