vitotai/BrewPiLess #268: Cannot upload esp32 with platformIO

tomohulk, 2020-10-04:
after a building the esp32 env, the upload fails:

```
Traceback (most recent call last):
  File "C:\Users\thomas\.platformio\packages\tool-esptoolpy\esptool.py", line 2959, in <module>
    _main()
  File "C:\Users\thomas\.platformio\packages\tool-esptoolpy\esptool.py", line 2952, in _main
    main()
  File "C:\Users\thomas\.platformio\packages\tool-esptoolpy\esptool.py", line 2652, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "C:\Users\thomas\.platformio\packages\tool-esptoolpy\esptool.py", line 222, in __init__
    self._port = serial.serial_for_url(port)
  File "c:\users\thomas\.platformio\penv\lib\site-packages\serial\__init__.py", line 88, in serial_for_url
    instance.open()
  File "c:\users\thomas\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '/dev/cu.SLAB_USBtoUART': FileNotFoundError(2, 'The system cannot find the path specified.', None, 3)
*** [upload] Error 1
```

Looks like a specific file is not being generated?

Win10 2004 x64 with latest VSCode/PlatformIO.

allthepies, 2020-10-04:
That's because the default platformio.ini config file is setup for a Mac.  You need to edit the file and change upload_port (and monitor_port if you want to debug) to the windows COM port the ESP device is connected to.  Try commenting out the upload_port entry first as platformio might detect the port for you.

tomohulk, 2020-10-04:
i was getting the same error on mac as well, i thought maybe it was something related to Big Sur, there is an issue with the esptool.py with Big Sur.

allthepies, 2020-10-04:
The /dev/cu.SLAB_USBtoUART port isn't guaranteed to work.  On my Mac sometime it works and sometimes the ESP pops up on a different device port.  You need to find out which port has been allocated to the ESP and change platformio.ini appropriately.  Try commenting out upload_port initially and see if platformio can find it automatically.

tomohulk, 2020-10-27:
finally getting around to messing with this.  But still no joy on Big Sur even when manually specifying the port:

```Linking .pio/build/esp32/firmware.elf
Retrieving maximum program size .pio/build/esp32/firmware.elf
Checking size .pio/build/esp32/firmware.elf
Building .pio/build/esp32/firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  15.4% (used 50304 bytes from 327680 bytes)
Flash: [========= ]  89.4% (used 1171294 bytes from 1310720 bytes)
esptool.py v2.6
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: /dev/tty.usbserial-0001
Uploading .pio/build/esp32/firmware.bin
esptool.py v2.6
Serial port /dev/tty.usbserial-0001
Connecting....

A fatal error occurred: Timed out waiting for packet header
*** [upload] Error 2
```

daghovland, 2020-10-28:
This is either a connection problem between your operating system and the board, or you have one of the boards that dont automatically go into flashing mode. Try f.ex. this: https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/

This is not an issue with the software here. I suggest you close this issue.

tomohulk, 2020-10-28:
I did try holding down the flash button with using VSCode and my Mac, and got the same error.  But using the NodeMCU Programer application on Windows works.  I guess until I need to customize it, its no big deal for now.