vitotai/BrewPiLess #269: Configuring http remote logging to Home Assistant

ksga, 2020-10-13:
Recently setup BrewPiLess on a Sonoff Basic R2 and got an iSpindel working with it.
I'm trying to enable remote logging to my Home Assistant setup.

It should be possible to simply create a "Long-lived token" and let BrewPiLess post a json formatted message to the API which should then create the entities.

Creating the token is fairly easy, but how to make BrewPiLess "consume" it?
According to the docs it should be possible to dousing curl: https://developers.home-assistant.io/docs/api/rest
I tried running:
`curl -X GET -H "Authorization: Bearer MYSECRETTOKEN" -H "Content-Type: application/json" http://HA_IP:8123/ENDPOINT`
from BPL_IP/testcmd.htm but it didn't seem to do anything.
Any suggestions on consuming the token?

Should I simply do it from another machine and upload it to the BPL filesystem somehow?

Afterwards - could the remote logging look something like this perhaps?
[image]

pedromargarido, 2020-10-13:
My suggestion for you is create a bridge using node red.

BPLess sends its data to a "http in" node on Node Red that will redirect it to Home Assistant.

ksga, 2020-10-13:
> 
> 
> My suggestion for you is create a bridge using node red.
> 
> BPLess sends its data to a "http in" node on Node Red that will redirect it to Home Assistant.

Sounds neat... if only I had a clue how to get started :)
I'm running Home Assistant in Docker, so the logical thing would probably be to create a node-red instance in docker as well. Will give it a try - but some pointers would be greatly appreciated.

pedromargarido, 2020-10-13:
I use HassIO, so I just installed Node Red as a Home Assistant add on. Just one click solved the problem. I'm not familiar with containers.

ksga, 2020-10-13:
okay.. starting node-red was fairly easy.
Now - setting it up will be the fun part ;)
Would it be possible for you to share your node-red flow, and BPLess logging configuration?

pedromargarido, 2020-10-13:
Brewpiless has MQTT communication, you can use this option to send data e receive commands directly from Home Assistant.

ksga, 2020-10-13:
not on Sonoff Basic which doesn't support the new ui :(

Hmm - seems the space between topic and message is a deal breaker for HA. Will look at it later.

vitotai, 2020-10-14:
Maintaining multiple versions of UIs is PITA. I am going to ditch support of Sonoff in the coming versions.

It's not difficult to support MQTT if flash space isn't a issue, though.

vitotai, 2020-10-14:
I tried a version with MQTT for SONOFF. You might try if you want.
https://github.com/vitotai/BrewPiLess/tree/Sonoff.mqtt