vitotai/BrewPiLess #270: Device settings lost after few minutes

flatplanetpl, 2020-10-15:
Hello,

I've was successful made adjustments and install latest version of brewpiless on sonoff S26. Relay and thermometers on 1wire works perfectly but...
after few minute all saved devices are gone. Even if I set erase eeprom still have same issue. Another thing is that I cant make backup any time with this configuration. I've got error 500.
Do you know what is en error related with? Sonoff s26 has 8266ex on board.

Thx for advices!

vitotai, 2020-10-16:
I haven't have a chance to get S26 on hand.
The error seems to be related with flash.
The 500 error is mysterious, though. Which configuration? The device configuration is "saved" by the browser, which has listed device information, while 500 is server "internal error".

flatplanetpl, 2020-10-16:
yes, those two issues are related. Sometimes I cant save new settings for relay and 1wire sensors. When I click "Init/erase EEPROM" after that I can save devices but still cannot make backup(error 500). Also without backup I can normally work with brewpiless only for couple minutes and all devices settings are gone and I have to do the same: init/erase and configure devices...

vitotai, 2020-10-19:
That's it! You are using wrong flash settings.
What is the other information in the system information panel?

closed 2020-10-23 by flatplanetpl