vitotai/BrewPiLess #276: Log version mismatched! BPL 4.01

Lichtus, 2020-11-12:
Hello all

I updated my BPL to 4.01 and after one day loging, there is an error with the info "Log version mismatched", I can't see any diagramm etc. and must start a new log file. I tried to do this several times. Did anyone have the same problem? Before I had a BPL 3.5 without any problems.

vitotai, 2020-11-12:
Reload the page. Force reloading.

meirion, 2021-01-02:
Same issue here.

V4.0 part way through a brew, changed the temp from 20c to 22c - the following day it started the
"Log version mismatched" error 

I've tried in a private browsing tab and clearing site settings but it still does the same in both.

Any suggestions? 
Thank you