vitotai/BrewPiLess #279: Heater power

looseends, 2020-11-14:
I'm using a 12v silicone pad heater with MOSFET using non inverted device. This heats the pad to 100C when heating.
Is there option to set power level via pwm to 30% would be ideal.

vitotai, 2020-11-14:
There is no such option. The latest official BrewPi seems to have one.

daghovland, 2020-11-15:
Really not an expert on PWM or mosfets, but if you always want 30%, would it not be possible, or even better, to just add a suitable resistor in series with the pad? Also, I would really recommend adding a fan to this setup, if you are able to. That would spread the heat out faster, and thus limit the heat increase in the pad.

pedromargarido, 2020-11-16:
The resistor would produce heat the same way the pad does. Actually the pad is a kind of resistor.

looseends, 2020-11-17:
At this stage I'll be feeding the output into pid heater on another esp8266.

vitotai, 2020-11-17:
IMO, a dimmer might be a good solution.