vitotai/BrewPiLess #282: ERR_CONTENT_LENGTH_MISMATCH - Graph doesn't display, sometimes pages don't display

meirion, 2020-11-25:
I have two brewpiless esp8622 boards.

One of them, always the graph does not display, and sometimes (often) the page will display as blank.

I've looked in dev tools and seen ERR_CONTENT_LENGTH_MISMATCH showing up whenever there is a blank graph/page

Wondering what is the cause and what I can do to fix, please?

[image]

[image]

vitotai, 2020-12-08:
First of all, make sure you hav correct file system size configuration by checking the sys info in "config" page.

I found this happens when the loading is too heavy or the power supply is insufficient. Moving the controller to a place near the AP/Router to see if it works.

One more possibility, you have a defect ESP8266. It happens.

meirion, 2020-12-08:
Thanks for the advice, I'll check those things.