vitotai/BrewPiLess #285: BPL 4.1: Restart loop after activating Brewafther

jacyker, 2020-12-09:
After upgrading to BPL 4.1, activating the Brewfather option causes the device to fall into a restart loop. Both for OTA and manual updates (including clearing all memory).

PS. It's nice if the update - especially OTA - did not delete the settings, for example remote log, system. Or maybe it is worth adding the import / export option for the settings?

vitotai, 2020-12-10:
My ESP32 runs well, but EPS8266 has issues.
Different issues happened with framework 2.6.2(can't connect to WiFi), 2.6.1(can't compile), and 2.6.0(Can't finish HTTP session). Still trying to find stable one.

jacyker, 2020-12-12:
4.1r2 fixed. THX :-)

jacyker, 2020-12-20:
And 4.1r3 back to restart loop :-(

By the way - maybe it is worth saving the settings, for example, to a json file and loading them on reboot, so that we do not have to set most of the update again?!?

vitotai, 2020-12-22:
I think I've found the "issue", which is the declaration order of HTTPClient and WiFiClient. 
It sounds stupid. I spent a lot time reading the source and check difference between 4.0 and 4.1 to find the only difference.

jacyker, 2020-12-22:
I confirm. It looks like it's working. I didn't even have to set anything this time (MQTT, BF ID).

THX 👍 

Best regards, Patryk

closed 2020-12-22 by jacyker

gieljnssns, 2020-12-26:
@vitotai 
I use v4.1 and I'm seeing the same when activating MQTT