vitotai/BrewPiLess #286: BPL 4.1: Not able to address others Controller PINs

acremonezi, 2020-12-10:
**Tested on:** ESP-32, 30-PINs.
Unable to address other available controller PINs other than 16, 17, 19, 26 and 27.

**How to reproduce the issue:**

Installed Device & Functions
- GPIO16 = Chamber Cooler
- GPIO17 = Chamber Heater
- GPIO19 = Chamber Door Switch
- GPIO26 = Chamber Light
- GPIO27 = Chamber Fan

**So, unable to address the following Device & Functions:**

- DHT Humidity Sensor
- Humidifier
- Deumidifier

vitotai, 2020-12-11:
It's designed intentionally.

Please read through this page: https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

Then, https://github.com/vitotai/BrewPiLess/blob/master/doc/ESP32Pins.md

You should understand the risk of using other free pins. If you want to try, understanding the risks, you can add the PINs by editing
line 202@DeviceManager.h
like
case 5: return 4; // GPIO4
case 6: return 13; // GPIO13

acremonezi, 2020-12-11:
Thank you very much!

closed 2020-12-11 by acremonezi