vitotai/BrewPiLess #287: Struggling to build for esp8266

iamandymcinnes, 2020-12-11:
I've just pulled the latest, and I've been through and updated everything platform and library, but when I try to build either the esp8266-dev project I just get ton's of errors:

Processing esp8266-dev (platform: espressif8266@~2.2.0; board: nodemcuv2; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (2.2.3) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 2.20502.0 (2.5.2)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa 2.100100.200706 (10.1.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 6.10.0
|-- <ESP8266_SSD1306> 4.0.0
|   |-- <Wire> 1.0
|-- <OneWire> 2.3.5
|-- <ESP8266WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP Async WebServer> 1.2.2
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 6.10.0
|-- <DHT sensor library> 1.4.0
|-- <ESP8266HTTPClient> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <Wire> 1.0
|-- <EEPROM> 1.0
|-- <ESP8266HTTPUpdateServer> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <Adafruit ADS1X15> 1.0.1
|   |-- <Wire> 1.0
|-- <PCF8574 - library for ESP8266> 1.0.0
|   |-- <Wire> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Building in release mode
Compiling .pio\build\esp8266-dev\src\Actuator.cpp.o
Compiling .pio\build\esp8266-dev\src\ActuatorArduinoPin.cpp.o
Compiling .pio\build\esp8266-dev\src\AutoCapControl.cpp.o
Compiling .pio\build\esp8266-dev\src\BPLSettings.cpp.o
Compiling .pio\build\esp8266-dev\src\BrewKeeper.cpp.o
Compiling .pio\build\esp8266-dev\src\BrewLogger.cpp.o
Compiling .pio\build\esp8266-dev\src\BrewPiLess.cpp.o
Compiling .pio\build\esp8266-dev\src\BrewPiProxy.cpp.o
In file included from C:\users\mcinnesa\.platformio\packages\framework-arduinoespressif8266@2.20502.0\tools\sdk\libc\xtensa-lx106-elf\include/stdlib.h:18,
                 from C:\users\mcinnesa\.platformio\packages\framework-arduinoespressif8266@2.20502.0\cores\esp8266/Arduino.h:27,
                 from src\Brewpi.h:62,
                 from src\Actuator.cpp:21:
C:\users\mcinnesa\.platformio\packages\framework-arduinoespressif8266@2.20502.0\tools\sdk\libc\xtensa-lx106-elf\include/sys/reent.h:405:11: warning: unnecessary parentheses in declaration of '_sig_func' [-Wparentheses]
[... 146 lines cut ...]
C:\users\mcinnesa\.platformio\packages\framework-arduinoespressif8266@2.20502.0\cores\esp8266/Arduino.h:271:12: error: 'isnan' has not been declared in 'std'     
  271 | using std::isnan;
      |            ^~~~~
*** [.pio\build\esp8266-dev\src\BPLSettings.cpp.o] Error 1
In file included from c:\dev\brewpiless\.pio\libdeps\esp8266-dev\arduinojson\src\arduinojson\operators\VariantCasts.hpp:7,
                 from c:\dev\brewpiless\.pio\libdeps\esp8266-dev\arduinojson\src\arduinojson\operators\variantoperators.hpp:7,
                 from .pio\libdeps\esp8266-dev\ArduinoJson\src/ArduinoJson/Array/../Variant/VariantRef.hpp:12,
                 from .pio\libdeps\esp8266-dev\ArduinoJson\src/ArduinoJson/Array/ArrayIterator.hpp:8,
                 from .pio\libdeps\esp8266-dev\ArduinoJson\src/ArduinoJson/Array/ArrayRef.hpp:9,
                 from .pio\libdeps\esp8266-dev\ArduinoJson\src/ArduinoJson.hpp:17,
                 from .pio\libdeps\esp8266-dev\ArduinoJson\src/ArduinoJson.h:9,
                 from src\BrewPiLess.cpp:17:
.pio\libdeps\esp8266-dev\ArduinoJson\src/ArduinoJson/MsgPack/MsgPackSerializer.hpp:28:7: error: attributes are not allowed on a function-definition
   28 |       ARDUINOJSON_NO_SANITIZE("float-cast-overflow") {
      |       ^~~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\esp8266-dev\src\BrewPiLess.cpp.o] Error 1
================================================================== [FAILED] Took 12.14 seconds ==================================================================

Environment    Status    Duration
-------------  --------  ------------
esp8266-dev    FAILED    00:00:12.141
============================================================= 1 failed, 0 succeeded in 00:00:12.141 ============================================================= 
The terminal process "C:\Users\mcinnesa\.platformio\penv\Scripts\pio.exe 'run', '--environment', 'esp8266-dev'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

daghovland, 2020-12-13:
Not sure, but there could be something wrong with the c++ version your comiler assumes. Hard to say without more details, f.ex. what compiler are you using (gcc?)  and how you are compiling this (PlatformIO?).

papatsit, 2020-12-15:
I have the same issue as "iamandymcinnes"
Can anyone help us ? I am not used also in PlatformIO . 
If i upload the hex file , is the Wifi Manager working ? Serial does not seem to respond .
Thanks .

vitotai, 2020-12-16:
Try "Clean" by clicking the trash can icon.