vitotai/BrewPiLess #288: Chamber Door

acremonezi, 2020-12-16:
Could you please inform/clarify if the "Chamber Door" actuator has any influence into the BPL logic?
Example: When the Chamber Door is open, is there any software logic that postpone any other action such as heater or cooling?
Could you please tell me where can I find this part into the code?
Thanks in advanced.

vitotai, 2020-12-16:
AFAIK, the state of door serves only two purpose:
1. event logging, maybe it might show on the chart of original BrewPi. (But not BPL)
2. light control

you can search, globally, "doorOpen", "isDoorOpen", "doorState", and "door" in the source. There isn't too much code.

I never thought of door even when I ran BrewPiless(legacy) on Arduino.
There are too few pins and I never thought of spare one for door.

acremonezi, 2020-12-17:
Thank you very much!

closed 2020-12-17 by acremonezi