vitotai/BrewPiLess #289: Install two DHT1x/DHT2x Sensors

acremonezi, 2020-12-17:
This is not an issue, it is only a doubt or maybe, if reasonable, an idea to improve and think about.

*Is it possible to install two DHT1x/DHT2x ?*

If it is possible, instead of using a Ds18b20 to measure **Room Temperature** and another one Ds18b20 to measure **Fridge Temperature**, we can use two DHT1x/DHT2x for that, one inside the fridge and another one outside and measure two information, Temperature and Humidity from the same sensor.

The second DHT1x/DHT2x to measure Room Temperatura, does not need to be involved on actuators control.

Best regards,

vitotai, 2020-12-18:
Only one DHTx is supported. One DHTx occupies one PIN. PINs are rare resource on ESP8266, and even on ESP32.

It is "possible" to implement, but current design is not flexible enough to support two DHTx.

BTW, the temperature reading of AM2301 seems sort of jumpy. I am not sure if it is good for temperature control.

PRyC, 2020-12-20:
Maybe it is worth thinking about BME?  Not only temperature, humidity and (atmospheric ;-)) pressure, but it is also much more stable... And it works on I2C, so more sensors don't mean more pins.

Yes, I know... The last approach to convince you (at least a little) ;-)

Patryk

acremonezi, 2020-12-21:
**@PRyC** 

BME, looks very interesting. Very well remembered!
The I2C capability is a good advantage due to the fact of the Micro-controllers pin restrictions.

**@vitotai** 
I am not 100% sure because I did not analyse the specs of both of tham, but I thing that the "jumpy" behavior you mentioned is probably due to the low hysteresis of the DHT against the Ds18b20 that we usually have it inside an stainless steel pipe. The DHT probably is faster on the measurement then the Ds18b20 due this hysteresis. you If keep the DHT and the Ds18b20 is a flow of air, the DHT will show moviments much faster tham Ds18b20.
As I know, humidity is temperature dependent.

I agree with you that the Ds18b20 is better for temperature control.

Best regards for all of you!

vitotai, 2020-12-21:
I've bought a few BME280.
It should be easy to integrate BME280 if the library works. The problem is the address.

BPL uses I2C LCD, and it's not convenient  for users to know the address of his I2C LCD, sometimes the sellers on eBay don't know, either. Therefore, BPL scan ALL I2C address to find one that response, assuming the only device is the I2C LCD. Introducing BME280 breaks this assumption. We will have to limit the range of addresses of I2C LCD.

acremonezi, 2020-12-22:
@vitotai

I understood you very well. You are far way ahead!
So, looks like BME has a good future against DHT due to I2C.