vitotai/BrewPiLess #290: ESP32- add a dedicated SSD1306 OLED to display pressure

Hwerow, 2020-12-19:
Hi Vito,
Thanks for all your good work. I am working on using BPL on an ESP32 to control my pressure fermentation. To avoid having an analog gauge I would like to display the actual pressure in psi on a second SSD1306 display. 
From my research, the ESP32 has the capability to define and use a second I2C connection which could use SDA pin 5, SCL pin 4 for a dedicated psi OLED display, which hardware-wise would be easy to add.
I have looked at the source code and I don’t have the programming knowledge to make this modification. 
Please would you consider adding this capability.

Thanks

vitotai, 2020-12-19:
I2C supports multiple devices. If the addresses are different, two LCD can be on the same I2C bus.
But, why not showing it at fifth line of OLED display?
I've thought of a customized 5th line display by specifying a printf-like setting.
eg: "Pressure: %P"  to display pressure, "Gravity: %g" for current gravity.

Hwerow, 2020-12-19:
Yes, I am aware that if you change the address on the OLED but it is very fiddly. I thought that the fifth line on the OLED was used for date and IP? I was thinking that a fixed and larger numerical display would be easier to see from a distance. Current gravity would be a useful value to have locally displayed as well.

Hwerow, 2020-12-20:
https://randomnerdtutorials.com/esp32-weather-station-pcb/

This project uses one SSD1306 and uses multiple screens that advance every 15 secs or by button push.  This might be a more elegant solution than having multiple OLED displays? It could achieve a dedicated screen to show the psi in a larger font which is what I am looking for. 
Cheers
Peter

vitotai, 2020-12-21:
The reason I keep the original BrewPi screen untouched is that I want to keep the user-interface, which is controlled by the rotary encoder.
It doesn't seem easy to comply the logics of original user-interface in the case. Otherwise, it's a brilliant idea.

BTW, to "draw" text on the OLED, "font" is necessary. I chose 6x12 font, so that it makes 128x64 display into 20 x 5. To display a larger text, the font data is needed. (Maybe only number is needed in this case.)

Hwerow, 2020-12-21:
So it would seem that using a second OLED for the  pressure information would be the simplest approach and the first OLED keeps the original BrewPi functionality with the rotary switch (which I have got working.) 
Yes, a number would work e.g. 20.5 psi I don't think more that one decimal place is required.

vitotai, 2020-12-23:
I kind of like the idea of flicking alternative display.
My initial thinking:
- change display every few seconds when idle
- if the rotary encoder is "turned", changed display
- if the rotary encoder is "pushed", change to primary display and enter menu mode

I haven't thought a "generic" secondary display, though. Including pressure, there are also gravity and humidity data. A flexible way is preferred.

Hwerow, 2020-12-23:
Sounds promising. If I understand correctly. Just the one OLED with different screens displayed with a configurable timer or by turning the rotary encoder. One of these screens could show pressure. Initiate the original BrewPi menu by a push of the encoder.