vitotai/BrewPiLess #293: ESP32

jsbjx, 2020-12-21:
BrewPiLess.v4.1r3.esp32.bin  Can anyone provide the module photos and wiring diagrams needed for this file? Thank you!

acremonezi, 2020-12-23:
Here it is what you need:

https://github.com/vitotai/BrewPiLess/blob/master/doc/ESP32Pins.md

Best regards

vitotai, 2020-12-23:
It's just struck me that if buzzer is moved to GPIO2, GPIO18 can be freed as general IO for more actuators or sensor.

acremonezi, 2020-12-23:
@vitotai 

Excellent idea!
After you informed me how to use other available pins I have been thinking about why not just define some "not so critical actuators" like chamber door, lights, fan and buzzer.

What do you think about the idea bellow?

**GPIO 0, 2 and 4 are Strapping Pins, it means each one is connected with its internal pull-up resistor during the reset.**

So, we could use then for:

**GPIO-0** to **Chamber Fan** (It has a PWM on boot, no worries on make some fan movements on boot, please see this [idea](https://github.com/vitotai/BrewPiLess/issues/250#issuecomment-749861558) I am thinking about it)
**GPIO-2** to **Chamber Lights** (just because it already have a LED connected on the board)
**GPIO-4** to **Buzzer** (It is HIGH on boot, so we will listen a BIP on boot)

**GPIO-34** to **Chamaber Door Switch** (Just becouse it is a "Input Only" available)

Best regards,

Hwerow, 2020-12-23:
Small point GPIO19 is used for Capping

acremonezi, 2020-12-26:
@vitotai 

Just tested this idea into a protoboard and it works very well.

[image]

acremonezi, 2020-12-26:
> Small point GPIO19 is used for Capping

@Hwerow I have not tested Capping yet, but as I understood, we can address it as we would like. There is not fixed GPIO for capping.
Example: OneWire temp sensor has a fixed GPIO defined, but chamber heater as well as capping does not have one, you can define any one free available.

Please, correct me if I am wrong, so I learn with it.

Best regards,

Hwerow, 2020-12-26:
Re the capping I guess I was too brief in my comment. It would have better read "What about the Capping?" which you have covered above. Cheers

acremonezi, 2021-02-14:
@vitotai 

Please, I would like to create a file "./src/CirculationFans.h" in order to firstly generate a fixed 50% PWM signal in any free ESP32 available IO and afterward implement this idea [here](https://github.com/vitotai/BrewPiLess/issues/250#issuecomment-749861558).

Could you please tell your recommendations and best practices in how to call this file? So, It can be usefull in the future?
I am analysing the code, but I was not able to recognize where should be the best practice.

Thanks in advanced.

Alcides