vitotai/BrewPiLess #294: Problem with saving devices

Manhem, 2020-12-22:
Hi.
I have tried your new software today and I have got one small problem.
When I shall save devices it does not work.
It's two I2C (DS18S20), heating and cooling.
I choose Device, clear eeprom, Refresh device list, choose Device slot and then press Apply.
This I do on every device. 
After this I try to take a Backup, I only get No installed devices.
If I go to another screen and the back, Refresh device list, nothing has been saved.

Any sollution on this problem??

vitotai, 2020-12-23:
DS18S20 uses OneWire instead of I2C.
You should "Refresh" before "backup".

Not enough information for me on your issue.

Manhem, 2020-12-23:
Hi.
I'm sorry, my fault.
There was one more thing I have to do that I missed.
Now it works.

closed 2020-12-23 by Manhem