vitotai/BrewPiLess #299: The appearance of the web page

WojtekGD, 2020-12-30:
The new version has a slightly broken page design:
[image]

Good job!
Greatings
Wojtek

WojtekGD, 2020-12-31:
I had to go back to version 4.0. Version 4.1 also has a problem with connecting wifi and receiving data from ispindel

vitotai, 2020-12-31:
> I had to go back to version 4.0. Version 4.1 also has a problem with connecting wifi and receiving data from ispindel

I had similar issues.
I upgraded my AP/router to a new WiFi6 one. The old versions worked fine, but versions with newer framework just couldn't connect. Sometimes, it connected but lost connection soon.
I changed the AP/Router and everything works as before.

I googled and found some interesting information. The new and replaced AP/router defaults to same SSID for 2.4G and 5G. However, the vendor suggests to use different SSIDs. Unfortunately, I returned the AP/Router soon, so I don't know whether or not changing to different SSIDs would work.  If you are in this condition, you might try.

There is also an DHCP issue of ESP8266 in AP_STA mode. If ESP8266 is trying to connect to network, DHCP service doesn't work on AP side. Therefore, iSpindle won't connect if ESP8266 is trying to recover WiFi.